robotology/mesh-support #678: Right shoulder goes in hardware fault during yoga

GiulioRomualdi, 2018-06-19:
### Description of the failure
Sometimes during the yoga the right shoulder goes in hardware fault; however, using the `yarpmotorgui`, we can restore the torque mode even if the the demo is not finished. 

### Detailed conditions and logs of the failure
Even if this issues does not occurs every time it is repeatable.
 Here you can find the logger  [yarprunlog_19_06_2018_16_01_11.log](https://github.com/robotology/icub-support/files/2115513/yarprunlog_19_06_2018_16_01_11.log)

hu-yue, 2018-06-21:
This happened in a quite systematic way during the demo on Tuesday, basically almost every time we ran the yoga, when the robot came to double support, the right shoulder went into hardware fault status. Errors are as the once from the previously log, we are not entirely sure if this is strictly related to the yoga (during the yoga the shoulders move a lot, differently from walking).

julijenv, 2018-06-21:
Hi @hu-yue ,
unfortunataley the log doesnt show anything about the right shoulder going in hardaware fault.
please snd me another log or double check that you sent me the good log.

DanielePucci, 2018-07-02:
@julijenv @spada1978 
May you help us with this? This is problematic now