robotology/mesh-support #680: right shoulder goes to hardware fault at startup of robot interface

nunoguedelha, 2018-07-04:
### Description of the failure
The robot interface startup fails as the right shoulder goes to hardware fault. Apparently it is due to a configuration/verification failure on the shoulder joint encoders. @julijenv had just replaced some cable in the encoder wiring, but the encoder is still failing. Not sure this is related to https://github.com/robotology/icub-support/issues/678.

### Detailed conditions and logs of the failure
The issue is systematic. Please refer to the attached log for more information.

nunoguedelha, 2018-07-04:
Emiliano Barbiera took a look this morning and checked that the joint connections are fine. The issue seems to come from the joint encoder (shoulder yaw) itself.

marcoaccame, 2018-07-04:
Hi, what Emiliano found is coherent with the the error type = 3 which is related to the chip.
```
typedef enum
{
    encreader_err_NONE                  = 0,
    encreader_err_AEA_READING           = 1,
    encreader_err_AEA_PARITY            = 2,
    encreader_err_AEA_CHIP              = 3,
    encreader_err_QENC_GENERIC          = 4,
    encreader_err_ABSANALOG_GENERIC     = 5,
    encreader_err_MAIS_GENERIC          = 6,
    encreader_err_SPICHAINOF2_GENERIC   = 7,
    encreader_err_SPICHAINOF3_GENERIC   = 8,
    encreader_err_AMO_GENERIC           = 9,    
    encreader_err_GENERIC               = 14,    
    encreader_err_NOTCONNECTED          = 15 /* this error happens when the encoder type is none or encoder is not local, for example it is connected to 2foc board */
} eOencoderreader_errortype_t;
```

julijenv, 2018-07-19:
Hi @nunoguedelha ,
could you lease close this isssue from what I recall I changed the internal sensor of the right shoulder joint 2. cheers

julijenv, 2018-07-19:
#678