robotology/mesh-support #683: Left lower leg skin board restarts

hu-yue, 2018-07-05:
### Description of the failure
The left lower leg skin keeps restarting some time after starting the robot with `yarprobotinterface`.

### Detailed conditions and logs of the failure

As we planned on using the soles for walking experiments, to check if the the skin of the soles are still working, we attached the soles to the robot. 
The first time the left sole was not working. To check if the problem was on the sole itself, we connected it to the right foot, and it was working. To investigate further, we connected the right sole to the left sole and we saw that it was working in the first minutes, then it stopped working. We saw that the left lower skin board was restarting (leds blinking at high frequency).
We then exchanged again the boards (right sole on right foot and left sole on left foot), and saw that the left lower leg skin board was indeed restarting. We also tried to attached the soles using CAN 12 instead of 13 but the behaviour was still the same, everytime we restart the robot, the left lower skin board restarts.

In the logger there is no error shown when this happens:
[log_icub-head_yarprobotinterface_7151_left_lower_skin_restart.txt](https://github.com/robotology/icub-support/files/2166643/log_icub-head_yarprobotinterface_7151_left_lower_skin_restart.txt)

cc @DanielePucci @fjandrad @nunoguedelha

DanielePucci, 2018-07-05:
CC
@marcoaccame @valegagge

marcoaccame, 2018-07-05:
I believe it is an electrical problem. Because on the right leg you write it works.

To help debugging this problem it is possible to see if the mtb boards are alive at startup of the device by adding the following xml code inside the `left_leg-eb10-skin.xml` file just before section `<group name="patches">`:

```
        <group name="SERVICE">
        
            <param name="type"> eomn_serv_SK_skin </param>

            <group name="PROPERTIES">
            
                <group name="CANBOARDS">
                    <param name="type">                 mtb                 </param>

                    <group name="PROTOCOL">
                        <param name="major">            0                   </param>    
                        <param name="minor">            0                   </param>     
                    </group>                    
                    <group name="FIRMWARE">
                        <param name="major">            2                   </param>    
                        <param name="minor">            17                  </param> 
                        <param name="build">            1                   </param>
                    </group>
                </group>
           
            </group>  
        
        </group>   
``` 

This xml code will force a discover of all mtb boards below the eb10 board by checking vs their presence and their compliancy to fw version 2.17.1. If a board is not present or not started or with a different fw version, then the log of yarprobotinterface will tell it and prevent the start of the robot.

nunoguedelha, 2018-07-06:
@marcoaccame , 
- **we actually are sure the boards are alive at startup**:
After the interface startup, we are able to see the contact detection in a yarp viewer while touching the soles.
- **the failure occurs later, apparently after a fixed amount of time following the startup**: 
in the case where we don't edit the xml file as shown above, can a wrong firmware version cause a delayed failure? like for instance some incompatibility leading to some buffer overflow, or memory violation leading to a restart?

marcoaccame, 2018-07-06:
As I wrote before, i think it is an electrical problem. 

I did not write the FW of the mtb, I only looked at it when I wrote the application for the new mtb4, but:
- I am not aware of any FW problem in mtb which causes a restart. 
- I can tell only that there is not a watchdog on mtb and any fatal error shall surely lead to a crash without restart.

davidetome, 2018-07-09:
We replaced both EMS4 **EB10** and MTB4 **10B10**. We left the robot up and running for a while (about 1 hour) and the problem seems to be solved; no restart happend and all skin pathces work properly.

closed 2018-07-09 by davidetome