robotology/mesh-support #686: Green_icub cameras not working on the head

mebbaid, 2018-07-10:
### Description of the failure

When
```
yarpdev --from camera/ServerGrabberDualDragon.ini --split true 
```
We face the following error:

[image]

### Detailed conditions and logs of the failure
This is within the context of running the Oculus with the head on the `iCubGenova04`
[icub_cameras.txt](https://github.com/robotology/icub-support/files/2179991/icub_cameras.txt)

mebbaid, 2018-07-10:
@traversaro if you can manage a 15 minutes, would be appreciated.

mebbaid, 2018-07-10:
Several reboots seems to do the trick !

closed 2018-07-10 by mebbaid

Nicogene, 2018-07-10:
For my personal experience it is firewire fault, unplug and plug it in the head or reboot usually does the trick!