robotology/mesh-support #687: Missing MTBs on EMS11

hu-yue, 2018-07-11:
### Description of the failure
When we attached the right skin sole, the logger saturated with many errors on EMS11, similar to those in https://github.com/robotology/icub-support/issues/685.

### Detailed conditions and logs of the failure
We attached the skin soles to the robot and the logger started saturating with errors on EMS11. Similar to https://github.com/robotology/icub-support/issues/685, we checked the FirmwareUpdater and saw that there were missing MTBs. This seems the reverse of #685, since in this case it happened when we attached the sole.

Log: 
[log_icub-head_yarprobotinterface_24639_skin11.txt](https://github.com/robotology/icub-support/files/2183191/log_icub-head_yarprobotinterface_24639_skin11.txt)

Screenshot of FirmwareUpdater:
[image]

marcoaccame, 2018-07-11:
There is the whole 10.0.1.11:CAN2 detached. The repair is: fix the cabling.

FirmwareUpdater cannot see board, yarprobotinterfae tells it cannot transmit on it. The latter situation leads to a huge number of problems due to the retransmission retries: delays, overflow of message buffers etc. That is the reason the logger saturates with errors.

hu-yue, 2018-07-13:
It has not happened again so far.
Closing for now, will open again in case it happens again.

closed 2018-07-13 by hu-yue