robotology/mesh-support #690: AEA encoder invalid data on right leg eb9 on iCubGenova02

FabioBergonti, 2018-07-20:
### Description of the failure
I was doing some jumping tests with the robot when this error occurred

```
6163	428,729995	ERROR	 from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 500s 167m 42u: (code 0x02000008, par16 0x0001 par64 0x000000000000000b) 
-> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

The robot was in contact with the ground, it jumped and then the motors have been switched off (probably due to an over-current of the board which commands the battery).
I've restarted the interface and every time that I launched it the right ankle remains in hardware fault.

### Detailed conditions and logs of the failure
Here below the logger
[log_icub-head_yarprobotinterface_2094.txt](https://github.com/robotology/icub-support/files/2214201/log_icub-head_yarprobotinterface_2094.txt)

hu-yue, 2018-07-25:
As @julijenv checked, it seems that it is the strain causing this issue.
For now the right foot strain has been disconnected and the robot is working fine.

fiorisi, 2018-08-02:
@julijenv we would like to use again the FT sensor on this robot. Do you think that we need to replace it?

julijenv, 2018-08-28:
Hi @FabioBergonti , @fiorisi , @fiorisi ,
I changed the 4 FT sensors of the legs and checked that every single boards of he legs are orking properly, please let eme know if it works!

julijenv, 2018-08-29:
Just as a reminder for any purpose that could be:
here is the SN of the different part of the legs respect to the previous configurations:
 upper left leg (SN 333 - prev. 222)       upper right leg (SN 334 - prev. 221)
 ankle left leg (SN 335 - prev224)          ankle right leg (SN 336 - prev. 223 -. this one was also disconnected to get further investigation as potentially defective)

fjandrad, 2018-09-08:
Thanks @julijenv you just save me a trip to IIT thanks to https://github.com/robotology/icub-support/issues/690#issuecomment-417021654 xD

closed 2018-09-08 by fjandrad