robotology/mesh-support #692: Left hand palm skin problem

gsaponaro, 2018-07-24:
Hi, we are not getting any values from the left hand skin palm part (the fingertips are OK).

We noticed that the red LED is not blinking, the orange LED (which we think should be on continuously) is off:

[image]

We have re-wired the power connections between the five small MAIS_FC.

We have tested the continuity between palm and MAIS_FC, it is OK.

**As a further test, we tried to connect the left palm skin to the right hand (see picture), and it works! In other words, this left skin part works when connected to the "other" right hand.**

Any other suggestions?

Is it possible (as a test) to physically swap the MAIS-L and the MAIS-R boards, but keeping the small MAIS_FC's as they are in the left hand? This would help us determine if the MAIS-L is actually broken.

/cc @rnun3s @vicentepedro

julijenv, 2018-07-25:
Hi @gsaponaro , @rnun3s, @vicentepedro ,
first thing first,
please remove the entire FC connectors from the MAIS board, and see if the MAIS boots when turn ON the switch for the motors. see if the LEDs do the same as the right one.
if not the MAIS board may have been messed up and you will have to reflash the firmware in it.
if  everything goes well, try to plug one FC connector at a time and do turn On and OFF until the MAIS dont boot correctly. -> it is possible that you have a little  short circuit somewhere please try to follow the wires along the fingers and the ones to the palm.
Or may be the real problem is a poor connection either from the power line going to the MAIS board  (check the voltage at the level of the MAIS input when you removed all the FC connector (do the same for the right hand and compare), you can also compare by adding one FC connector at a time and compare the drop of voltage in between left and right hand, if the left is lower than the right please consider re crimping the power line of the MAIS either at the level of the direct input of the MAIs or at the level of the filter or higher on the upper arm.
let me know

gsaponaro, 2018-07-25:
Update:
* we removed the FC connectors, LEDs initially they were wrong (red not blinking, orange off), but we reflashed the firmware (mais.hex) and **now LEDs are ok**
* power line: we have re-wired the power connections between the five small MAIS_FC
* we re-built all the wires from the FC board where the palm wires connect
* we measured the voltage on the 4 wires of the left palm, they are similar to the right palm (3.3 V supply black&red, ~3 V green&white)

Further tests that we did:
* we tried to swap the wires N5 and N6, the problem persisted: fingertip skin OK, palm skin not working
* we also swapped the MAIS_L and MAIS_R, the problem persisted
* as in the first report, we connected the left palm skin to the right hand, and the left palm skin works (appearing as "right skin" in the yarp port / skin GUI, of course)

Note: in the last few years we never really used the skin, we want to use it now, it is possible that this problem was present from before.

Any other suggestions?

julijenv, 2018-07-30:
Hi @gsaponaro,
1- could you make sure that the 4 wires of the palm which got soldered on the FC connector are not broken.
2- Could you make sure that all the wires above the 2nd FC connector from the thumbs (where there is also the palm wires soldered) are similarly wired.
3- Could you make sure that the connector in between the cable of the palm and the FC, the wires and wiring are similar with the right hand.

gsaponaro, 2018-07-31:
Hi @julijenv , do you have any slot to Skype today?

julijenv, 2018-07-31:
Hi @gsaponaro, unfortunately il on holiday, il be back on next tuesday

gsaponaro, 2018-07-31:
OK, enjoy :)
I won't be here, my colleagues will look for you.

vicentepedro, 2018-08-22:
Hi @julijenv, do you have any slot to Skype during this week?

julijenv, 2018-08-22:
Hi @vicentepedro  and @gsaponaro ,
Im right behind skype if ever needed.
let me know

vicentepedro, 2018-08-22:
Hi @julijenv,
can we schedule a skype tomorrow at 3pm (Italy time)?

julijenv, 2018-08-23:
Hi @vicentepedro ,
yes 3pm italy time is ok for me

vicentepedro, 2018-08-24:
Hi @julijenv 
We found and solved the problem. The palm is working now.
There was a short-circuit between the SDA3 and VCC in one of the MAIS boards.

closed 2018-08-24 by vicentepedro