robotology/mesh-support #693: Right hand thumb seems broken

GiulioRomualdi, 2018-07-25:
### Description of the failure
During a walking experiment the right hand impacts the right leg and the thumb seems broken. 

### Detailed conditions and logs of the failure
In the following a `gif` is attached. In details:
1. we put the finger in idle
2. we run the finger
3. at the first touch the finger is closed  
4. we cannot move from the `yarpmotorgui` 

[image]

cc @hu-yue  @mebbaid

julijenv, 2018-07-30:
it has been restored.
IITCODE  Qta Description
1815 x 1m : cable carlstahl 0.63mm non coated

closed 2018-07-30 by julijenv

GiulioRomualdi, 2018-07-30:
Thank you @julijenv !