robotology/mesh-support #695: neck cable broken

amaroyo, 2018-08-01:
The neck cable if the iCub broke yesterday. Please take a look to the red arrows in the picture. The simetrical cable seems alright. 

[image]

spada1978, 2018-08-01:
Il cavo rotto in oggetto e cavo B.
vengo nel pomeriggio ad effettuare la riparazione, in linea di massima per le 14:30 al massimo alle 15

amaroyo, 2018-08-01:
perfetto! Grazie mille!

spada1978, 2018-08-03:
ho effettuato la sostituzione del cavo B del collo ma durante la lavorazione si è verificato il problema tra la puleggia e il supporto porta cavi.
e stato chiesto supporto a Torrazza quindi si procederà con la sostituzione del pezzo non appena la commessa di officina sarà pronta.

julijenv, 2018-08-10:
it has been replaced also the cable A of the neck after being a bit worn out.
I checked both motor and the ballbearing part of the harmonic drives , I remove the dust from one, and consequently check the moving of the neck entirely
I couldnt find any problem related to the neck afterwards.
IITCODE Qta Description
5965 x 1 : HEAD CABLE A

closed 2018-08-10 by julijenv