robotology/mesh-support #699: Skin not starting on iCubGenova02

traversaro, 2018-08-07:
### Description of the failure
yarprobotinterface fails to launch if the skin is enabled. If all the skin devices are disabled in the yarprobotinterface configuration, it is working fine. 

### Detailed conditions and logs of the failure
Launching with `icub_wbd.xml`, I get this error:
~~~
1772	61,009260	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1774	61,009323	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.5 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1779	61,506335	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1781	61,506407	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.5 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1784	62,013684	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1786	62,013806	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.5 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1788	62,013856	ERROR	EthResource::serviceCommand() failed an acked activation request to BOARD torso-eb5-j0_2 with IP 10.0.1.5 after 3 attempts each with waiting timeout of 0.5 seconds
1790	62,013915	ERROR	embObjMotionControl::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD torso-eb5-j0_2 (IP 10.0.1.5)
1858	62,611540	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb1-j0_3 IP 10.0.1.1 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1860	62,611628	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.1 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1864	63,108772	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb1-j0_3 IP 10.0.1.1 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1866	63,118942	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.1 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1870	63,595282	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 8s 955m 173u: (code 0x06000004, par16 0x0000 par64 0x0000000000000000) -> ETH monitor: just started. + .
1872	63,595356	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 8s 955m 292u: (code 0x00000009, par16 0x0000 par64 0x0000000000000000) -> SYS: the board is running happily. application state is in par16&0x000f : 0 -> just (re)started, 1 -> idle, 2 -> running. + .
1876	63,605402	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 8s 966m 171u: (code 0x06000004, par16 0x0000 par64 0x0000000000000000) -> ETH monitor: just started. + .
1878	63,605476	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 8s 966m 291u: (code 0x00000009, par16 0x0000 par64 0x0000000000000000) -> SYS: the board is running happily. application state is in par16&0x000f : 0 -> just (re)started, 1 -> idle, 2 -> running. + .
1880	63,615558	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb1-j0_3 IP 10.0.1.1 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1882	63,615627	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.1 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1884	63,615686	ERROR	EthResource::serviceCommand() failed an acked activation request to BOARD left_arm-eb1-j0_3 with IP 10.0.1.1 after 3 attempts each with waiting timeout of 0.5 seconds
1886	63,615745	ERROR	embObjMotionControl::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD left_arm-eb1-j0_3 (IP 10.0.1.1)
1933	63,696727	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 9s 59m 139u: (code 0x06000000, par16 0x0000 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1935	63,696795	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 9s 59m 284u: (code 0x06000000, par16 0x0001 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1937	63,696851	ERROR	 from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 9s 59m 423u: (code 0x06000000, par16 0x0002 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1941	63,696961	ERROR	 from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 8s 975m 165u: (code 0x06000004, par16 0x0000 par64 0x0000000000000000) -> ETH monitor: just started. + .
1943	63,697017	ERROR	 from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 8s 975m 280u: (code 0x00000009, par16 0x0000 par64 0x0000000000000000) -> SYS: the board is running happily. application state is in par16&0x000f : 0 -> just (re)started, 1 -> idle, 2 -> running. + .
1945	63,717052	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 9s 69m 41u: (code 0x06000000, par16 0x0000 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1947	63,717123	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 9s 69m 186u: (code 0x06000000, par16 0x0001 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1949	63,717178	ERROR	 from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 9s 69m 325u: (code 0x06000000, par16 0x0002 par64 0x3000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1952	63,808342	ERROR	 from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 9s 77m 374u: (code 0x06000000, par16 0x0000 par64 0x1000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1955	63,808532	ERROR	 from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 9s 77m 513u: (code 0x06000000, par16 0x0001 par64 0x1000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1956	63,808565	ERROR	 from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 9s 77m 652u: (code 0x06000000, par16 0x0002 par64 0x1000000000000000) -> ETH monitor: link goes up in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal). Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
1958	64,163410	ERROR	theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb2-j4_15 IP 10.0.1.2 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
1961	64,173651	ERROR	theNVmanager::Impl::command() fails a wait() from IP 10.0.1.2 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
3138	74,931811	ERROR	One or more devices failed opening... see previous log messages for more info
~~~

`icub_all.xml` is not even starting, because it gives this error: 
~~~
[... cut ...]
8919	641,595187	WARNING	Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGenova02/./icub_all.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV1.0
8920	641,595208	DEBUG	yarprobotinterface: using xml parser for DTD v1.x
8921	641,595220	DEBUG	Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGenova02/./icub_all.xml
8924	641,595819	FATAL	Syntax error while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGenova02/./icub_all.xml at line 5 . Expected "param", "group", "paramlist", "subdevice", or "params". Found xi:include
8929	641,604668	INFO	 Execution terminated
~~~

traversaro, 2018-09-12:
@gabrielenava @nunoguedelha this issue is still present I guess, right?

nunoguedelha, 2018-09-12:
I think not. **I can check this afternoon**, but if I'm not mistaken it has been fixed and the respective workaround disabling them has been cleaned-up in `robots-configuration`. @fiorisi ?

gabrielenava, 2018-09-12:
Cannot access the iCubGenova02 room at the moment, but I looking at the startup script [icub_wbd.xml](https://github.com/robotology/robots-configuration/blob/iCubGenova02_V2-5_plus_soles/iCubGenova02/icub_wbd.xml#L56) in the `robots-configuration/iCubGenova02_V2-5_plus_soles` branch it seems the skin (besides the torso) is enabled. I do not remember if this file has been edited locally.

julijenv, 2018-09-12:
Hi uys, @nunoguedelha , @gabrielenava ,
could you just close this isssue if the issue has been repaired?!?!?
thx in adfvcane