robotology/mesh-support #702: Neck broke due to falling

mebbaid, 2018-08-14:
### Description of the failure
While teleoperating the robot walking and grasping using the Oculus and the Virtualizer platform, the robot fell on its back.

### Detailed conditions and logs of the failure
Possible reasons would be battery dying out, and fault line stuck.
The following video demonstrate the event

[image]

### Note
We are not sure if the only part broken is the neck. It is still to be verified.

@hu-yue

hu-yue, 2018-08-16:
After @julijenv repaired the neck, the robot is working fine, we have tested several things and no issues arose.

However, it has to be noted that the battery that was inside (the "green-ish" one) has broken during the fall, in particular it seems that the BMS is broken, apart from some part of the plastic holder. Few hours after the fall, the BMS was very hot and still today (two days after) it was warm. @MrAndrea is taking a look at it.

MrAndrea, 2018-08-21:
I checked the battery and I found wires broken under the BMS board, and short circuits between some Lithium cells and BMS. I repaired the wires but the BMS seems burned.

[image]
[image]
[image]
[image]
[image]

MrAndrea, 2018-09-13:
I have the offer from the supplier regarding a new BMS, and I'm waiting for it.

hu-yue, 2018-10-25:
The robot fell again, and we broke another battery. The status seems a bit better than the previous one. The battery has been handed to @MrAndrea. 
Now we have two batteries left, one with working LEDs and one with not working LEDs. The one with not working LEDs cannot be housed properly in the battery pack, i.e. it had connection issues and we could not use it eventually. We are not sure if this might be due to the housing deformed after the fall or the battery itself, or both.

hu-yue, 2018-10-25:
cc @DanielePucci, @S-Dafarra, @GiulioRomualdi

DanielePucci, 2019-01-25:
Thanks to @MrAndrea and @maggia80, batteries have been repaired. We currently have three working batteries

closed 2019-01-25 by DanielePucci