robotology/mesh-support #703: Network conflict between Ethernet cable and WiFi on icub-head

hu-yue, 2018-08-16:
### Description of the failure

When the cable is plugged and the wifi is also on, often the connection with the head is lost.

### Detailed conditions and logs of the failure

From what we understood, in theory the cable and wifi should not go in conflict, i.e. if the cable is plugged in then it should prioritize the cable.
However, as described above, the conflict happens relatively often, and sometimes the connection is re-established in few seconds, sometimes it doesn't until one of the two connections is turned off (so ethernet unplugged or wifi switched off from the robot side).
We have been experiencing this problem since some time, it is difficult to debug because we do not know when it happens, and many times it happens for just a short time.

Also, with the wifi, sometimes the connection to the head is lost, for few seconds or longer.

@DanielePucci @nunoguedelha @mebbaid @GiulioRomualdi @S-Dafarra FYI.

nunoguedelha, 2018-08-16:
For those who don't need the wifi, but rather a stable wired connection, there is a workaround for avoiding the issue, while we wait for a fix: turn off the wifi service/device on the head, using the `ifconfig` command on the terminal as shown below:
```
sudo ifconfig wlp2s0 down
```
the interface name `wlp2s0` can be found by typing `ifconfig` in the terminal. The service will be active again as soon as you restart the icub head.

mbrunettini, 2018-08-23:
Please note that **the above "workaround**" : 
`sudo ifconfig wlp2s0 down`
actually breaks the network bond configuration and then **must not be used**

I am checking the issue, but since now I am unable to replicate it

mbrunettini, 2018-08-23:
I can spot 2 issues above:
1.  > When the cable is plugged and the wifi is also on, often the connection with the head is lost.

_I tried to connect and disconnect the cable several different times, but I was unable to replicat ethe issue_

2.  > Also, with the wifi, sometimes the connection to the head is lost, for few seconds or longer.

_I tested the wifi with several iperf sections of 7200 seconds with constant full bandwith usage and the connection has never lost_

My idea is that at least the second issue is triggered by hi CPU usage or overheating during walking experiment sessions. Regarding the cable plugging issue, it can be related since I was unable to replicate it. Moreover i have to say that **it is safer to perform the cable-wifi switch with low bandwith usage** (i.e. with now running process on the icub-head network)

Since the COM express board is going to be replaced, I proposed to close this issue and **see if it happens again with new hardware**: in such case, I will need to replicate the issue during experiment session (in real conditions), so please take note of setup state when issue is facing (which software was running, robot location, uptime of icub-head, etc), so we can replicate it, thanks.

closed 2018-08-23 by mbrunettini