robotology/mesh-support #707: Start-up failure due to communication errors on left leg boards.

ahmadgazar, 2018-08-28:
### Description of the failure
iCub is failing to start-up  due to communication erros on the left leg boards as stated here:

[image]

We restarted the robot a couple of times, however the error remained the same.
@julijenv 

### Detailed conditions and logs of the failure
Details can be found on the attached log file.
[log.txt](https://github.com/robotology/icub-support/files/2327829/log.txt)

cc: @fjandrad

marcoaccame, 2018-08-28:
@julijenv: the CAN cable on EB6:CAN2 is either unplugged or broken. The EB6 board cannot send CAN message to the STRAIN and complains w/ the above messages.

ahmadgazar, 2018-08-28:
Exactly @julijenv fixed that. Thanks.

closed 2018-08-28 by ahmadgazar