robotology/mesh-support #709: Cannot Run YarpRobotInterface on iCubGenova02

FabioBergonti, 2018-08-30:
### Description of the failure
I've tried to run yarprobotinterface to do some tests but the Robot didn't start
I've switched off the motors and the cpu, I've repeated the operation but the behavior was the same

### Detailed conditions and logs of the failure
[log](https://github.com/robotology/icub-support/files/2335906/yarprunlog_30_08_2018_13_02_25.log.txt)

cc @gabrielenava

julijenv, 2018-08-30:
The erro message is 

> WRONG BOARD TYPE

As the robot newly got strain2 F/T sensor mounted I guess the error is due to this.
AS @marcoaccame , suggested please refer to the config files of the robot iCubgenova04 the GREENCH and changed opportunately those files in order t oreference the good config of your robot meaing you use strain2 instead of the old STRAIN.
let us know

valegagge, 2018-08-30:
I verified that the problem is due to wrong configuration for board strain2.

@gabrielenava and I updated those files (see https://github.com/robotology/robots-configuration/commit/982beeaae50c245cda0f41bd96f7bb97df99a5de )  and the robot works properly.

closed 2018-08-30 by valegagge