robotology/mesh-support #716: iCubGenova02: Broken left hand wrist plastic

Yeshasvitvs, 2018-09-14:
### Description of the failure

### Detailed conditions and logs of the failure

hu-yue, 2018-10-05:
It was glued and now it's broken again.
We arranged a temporary "arrangement" to use the robot without breaking further the wrist.
I think the palm cover has to be replaced, maybe we can order the pieced and replace it, also considering that the robot has been scheduled to be off for about two weeks starting from the 10th October. 

cc @fiorisi @FabioBergonti @julijenv 

See pictures below:
[image]
[image]

Yeshasvitvs, 2018-11-19:
@hu-yue I am assuming this is fixed. Please close the issue if this is no longer required.

closed 2018-11-19 by julijenv