robotology/mesh-support #722: Overcurrent on EB9 and i2t limit exceeded on EB9

hu-yue, 2018-09-20:
### Description of the failure

During the Yoga experiment, we experienced first the an overcurrent on EB8 (joint 0, so ankle pitch), and then i2t limit exceeded on EB9 (joint 1, so hip roll).

### Detailed conditions and logs of the failure
Log here: 
[log_icub-head_yarprobotinterface_1952.txt](https://github.com/robotology/icub-support/files/2401046/log_icub-head_yarprobotinterface_1952.txt)

Could the issue on EB9 be related to #719?

julijenv, 2018-09-20:
Hi @hu-yue , is it still occuring?!?!?

julijenv, 2018-09-27:
did you test it if it was due to different branch work and consequently current limit changes?
if yes please close the isssue thx in advance
@hu-yue

hu-yue, 2018-10-01:
This issue has been solved after @gabrielenava changed the positions of the yoga, it seems it was pushing towards the limits. Closing.

closed 2018-10-01 by hu-yue