robotology/mesh-support #728: Left Hand failure

lrapetti, 2018-10-01:
When I try to run the yarprobotinterface, it crashes due to the following left-arm failure:

```
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15) @ 144s 713m 458u: CAN discovery after 1121 ms has detected 1 missing eobrd_mais boards in CAN1:
[ERROR]1 of 1: missing eobrd_mais BOARD 10.0.1.2:CAN1:14
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 144s 713m 573u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000500) -> CFG: EOtheMAIS cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 144s 713m 686u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000500) -> CFG: EOtheMAIS cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15) @ 144s 713m 832u: CAN discovery after 1121 ms has detected 1 missing eobrd_mais boards in CAN1:
[ERROR]1 of 1: missing eobrd_mais BOARD 10.0.1.2:CAN1:14
```
Here you can find the log [log_leftHand.txt](https://github.com/robotology/icub-support/files/2434459/log_leftHand.txt)

fiorisi, 2018-10-02:
This morning D. Gandini has checked the wires. Apparently there were no issues and now the robot seems to work (we run the yarprobotinterface 3 times without any error). Probably a connector was not plugged properly.  
We are going to do some test in the afternoon and close the issue if the error doesn't manifest again.

lrapetti, 2018-10-04:
This morning the error showed up again

closed 2019-10-12 by pattacini