robotology/mesh-support #730: Shaking when putting the robot in home pose

pi-q, 2018-10-11:
Hi! 

Since a few days, we have noticed that when using yarpmotorgui to bring the robot to some home pose, it results in the robot shaking and errors appear in yarplogger, see attached log:
[log_pc104_yarprobotinterface_1550_shakinghomepose.txt](https://github.com/robotology/icub-support/files/2469316/log_pc104_yarprobotinterface_1550_shakinghomepose.txt)

Stabilizing the legs by hand is sufficient to stop the shaking, and get back to normal. We took a video so you can understand better what I mean: https://drive.google.com/file/d/1qgH_-uEIXPPVA63WyQY0WJAhIHwyZi_y/view?usp=sharing

@pencoj1 tried to check connections in the back of the iCub but that didn't solve the problem.

Any idea what this could be coming from?

I won't be around to follow the issue, but @pencoj1, @InriaBrice should be there.

pi-q, 2018-10-17:
I am not convinced about the hypothesis of the FT sensor saturations. As far as I know, we have been having similar errors for more than a year, but that never caused that kind of shaky behaviour.

For sure there has been no PID changes lately, so then the next time that someone uses the robot, the screws will be checked.

As mentioned above, I won't be around, but @lpenco and @InriaBrice can eventually provide follow up.

pencoj1, 2019-03-11:
Hey Julien,
So the problem is in the hip roll of the left leg. When the robot is moving to fast the hip roll value start oscillating in a range of 3deg.
The PID values of left and right leg are the same.
Can I have a sketch of the connections I have to check with the wires numbers? like the one you sent me for the torso last time.

julijenv, 2019-03-11:
Hi @pencoj1 ,
After having taken a look again at the video, it looks quite clear to me that it is due to some loose screws.
can you double check that the screws on this picture from CAD are well tight.
[image]
Does the shaking restart if you push left leg and release it?
Does the left leg as more play than the right one? (test from calibration position just try to move the leg and see if you have more play).
You can also try to catch me on skype to make some more tests to see if it isnt some else inside this motor, i doubt that it could be something related to the electronics though.
Let me know

pencoj1, 2019-03-11:
Ok.The screws are tight.
 the shaking does restart if I push the left leg. I can't check if it has more play because if I try to move open the leg,(increasing by the hand the hip roll) it starts shaking.

pencoj1, 2019-03-14:
I had to chance to try again now. it is still shaking when I move fast the hip roll. 

This time tho I tried to move fast the hip roll of the right leg and that triggered the shaking. Then the right hip (pitch and roll) goes in hardaware fault. 

So I'm not sure if the problem is in the right or the left leg or both. 

When it starts doing it both the hip roll start shaking but before I thought it was the left, maybe they are both.

In anycase if tomorrow you'd be up for a skype, it'd be great to solve it hopefully once for all

pencoj1, 2019-03-18:
Sorry I forgot to attach the log of the errors I got for the hips
Here it is

[image]

julijenv, 2019-03-18:
Hi @pencoj1 ,
could you please use the tab LOG to perform a full copy of the logfile instead of a picture?!?
thx in advance,

pencoj1, 2019-03-18:
[LUIGI_HIP.log](https://github.com/robotology/icub-support/files/2978042/LUIGI_HIP.log)

julijenv, 2019-03-18:
Hi @pencoj1 ,
from the video and the log it is not quite clear what is going on,
the log file says that the FT sensor got rubish values and probably just prevent the robot from getting broken by putting it into Hardware Fault, could you double check this on yarpmotorgui , if you repaet the same experiment or behaviours.
let me know
Im right back my desktop on skype if you want to give me live feedback

pencoj1, 2019-03-18:
Just tried with left leg off.
if I move the hip roll of the right leg very fast, it doesn't start shaking and it doesn't go in hardware fault.
I just get a saturation of torque error which I think is normal since I 'm moving the leg very fast.

[log_pc104_yarprobotinterface_1135_ONLY_RIGHT_LEG.txt](https://github.com/robotology/icub-support/files/2978931/log_pc104_yarprobotinterface_1135_ONLY_RIGHT_LEG.txt)

pencoj1, 2019-03-19:
Hey Julien. I unscrewed all the screws and I found that 2 of them where broken. I removed those fucking bastards broken screws, I changed all of the other screws and now the leg is well tight. It appears that the problem is solved. I tested on the pole and it's fine. 
I'll consider the problem completely solved when I'll test the walking after I fix the left arm.

I'll change the screws also on the other leg to be sure.
thx a lot for the great support!

pencoj1, 2019-03-26:
Ok it's still giving the error. Right hip pitch and roll go in HW fault when I stress the hip roll of the robot
The screws are tight 100% now
Timestamp 1810
[LUIGI_HIP.log](https://github.com/robotology/icub-support/files/3009564/LUIGI_HIP.log)

pencoj1, 2019-03-27:
The hips they go in saturation also when I stop my controller as you can see in the video. If I kept moving the robot in such case then the right hip would go in HW fault.
[VideoLog.zip](https://github.com/robotology/icub-support/files/3012524/VideoLog.zip)

Is there a way to avoid this HW fault? because in this case the robot is not vibrating or doing dangerous things

pencoj1, 2019-03-27:
Now I calibrated the robot without FTs but when I control the robot to move the hip rolls, the right hip goes in HW Fault. 
This HW Fault seems to happen right after a vibration on the left hip as you can see in the video.
 The vibration this time stops soon because the screws are tight but it's enough to cause the HW F of the other leg.
PS Sorry I'm texting too much but I'm trying to give you a better understanding of the problem
[VideoLog1.zip](https://github.com/robotology/icub-support/files/3012698/VideoLog1.zip)

julijenv, 2019-03-27:
Hi @pencoj1, please reach me tomorrow on akyoe, today I'm away from workplace.
Cu

julijenv, 2019-03-28:
Hi @pencoj1 ,
I wet over the overall log and there wasnt any HW FAULT to be detected, strangely enough?!?

pencoj1, 2019-03-28:
Here it is a bunch of yarpscopes. I reported the yarpscope of the right_leg, left_leg, right_foot and left_foot before in the air and then putting the robot on the ground. The transition you see in the plot is when I indeed land the robot on the ground. 
You can also see the video when the right hip pitch and roll go in HW fault. At the end of the video the last vibration triggers the HW fault. You can see a yarpscope of the forces on the right leg during this HW fault as well.  
You can see in MoreYarpscopesSameTime folder that if I open 2 yarpscopes at the same time the overlap in the same yarpscope making it oscillate.
Legend: (Fx red, Fy green, Fz blue, Wx white, Wy yellow, Wz black)   
[LOG.zip](https://github.com/robotology/icub-support/files/3018630/LOG.zip)

julijenv, 2019-03-28:
@marcoaccame , @S-Dafarra , @gabrielenava , @traversaro , COuld you please take a look at those scopes and tell me if the right_leg FT could not have a problem (the Force at rest inside the FT is over 160N, isnt it too much)?
let us know

gabrielenava, 2019-03-28:
cc @fjandrad

serena-ivaldi, 2019-03-29:
I can see in the log this message that is quite weird (not signaled as error)

``In  right_leg-eb9-j4_5  there isn't 2FOC.Verbose filed. For default it is enabled 
<#STRING_END#>
737
10,155915
4
<#STRING_START#>AutoCalibration parameter not found for board right_leg-eb9-j4_5in bottle"2FOC" (HasHallSensor 1 1) (HasTempSensor 0 0) (HasRotorEncoder 1 1) (HasRotorEncoderIndex 0 0) (HasSpeedEncoder 0 0) (RotorIndexOffset 0 0) (MotorPoles 8 8) 
<#STRING_END#>
738
10,155934
4
<#STRING_START#>In  right_leg-eb9-j4_5  there isn't 2FOC.AutoCalibration filed. For default it is disabled 
<#STRING_END#>
739
10,186135
2
<#STRING_START#> from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 24s 826m 469u: (code 0x05000015, par16 0x2000 par64 0x0000000000ff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .``

pencoj1, 2019-03-29:
I reran a few experiments and I finally understood that the shaking is triggered by the right leg.

 It makes more sense now, since it's the right that then goes in HWFault. Due to that shaking, which is not visible when the robot is placed on the ground but it's there, the motor goes in HW fault.

 It's difficult to understand what causes the shaking since there are no errors in the log except for the FT strain saturation values, that according to Marco would not be the cause. Here I print everything, also debug, but I see no problems honestly. Once I ran without left leg and once without right leg. With the left leg only, there's no vibration. With the right, there's vibration and HW fault.

[LOG_OneLegOnly.zip](https://github.com/robotology/icub-support/files/3022730/LOG_OneLegOnly.zip)

julijenv, 2019-04-01:
Hi @serena-ivaldi and @pencoj1 ,
I double check and effectively the log file has mentionned 

> In right_leg-eb9-j4_5 there isn't 2FOC.Verbose filed. For default it is enabled

but it has this not only for the right_leg-eb9_j4-5 but also for right_leg-eb8-j0-3, left_leg-eb6-j0-3 and finally for left_leg-eb7-j4-5. Therefore I doublt that the problem hat you have is related to this WARNING messages.
@marcoaccame , @valegagge Could you please have a moment to check the log file attached to the last reply from @pencoj1 and tell me if you find any strange messages to help me to debug this issue?
In any case does the saturation on the FT sensor could lead to the HW FAULT?

To @pencoj1 , 
> If you guys have no idea of the problem, and since there are no errors in the log, I think we could try to change the right motor, at this point. what do you think?

I dont think the motor is broken in any possible ways, something trigger the HW FAULT but I dont think it is the motor. We may try other way to test it, I will wait for getting some eedback from the FWists and it could most probably enlightus on what could affect your robot.

julijenv, 2019-04-01:
Hi @pencoj1 ,
I recheck the final good quality video that you added to the yarpscopes and I from that It is quite clear that something is sliding in the hip - Can you please, block with 4 screws the robot on its stand , make sure that all the screws (the ones that you change are perfectly screwed for both legs) because the external ring move it should not be possible.
I am behind skype if ever you need me

julijenv, 2019-04-01:
Hi @pencoj1 
from our problem it looks like tose screws are loose.
[image]
I'm going to explain how to retighten them

julijenv, 2019-04-01:
HI @pencoj1  and @serena-ivaldi ,
We finally pinpoint after a skype meeting the real loose part of the legs. 
The problem now is a matter of tightening loose screws (as expected). the issue is how to do so, becasue it is quite an advanced level task. WCS scenario if done with care is recalibrating the joint.
I'm gonna try to explain step by step how to do.
First thing first,
disassemble the lower of the leg by removing the 8 screws M4x10
[image]
leave the leg along the stand attach with some tie wraps if you have or whartever could hook it up along.
Unmount green shape by removing the 8 screws M3x8
[image]
unmount the outer ring by unscrwing the 8 screws M3x25
[image]
slide off the grey ring gently
[image]
All the uncircle pieces should come went you gonna slide off fthe grey piece.
At that point you should be able to understand with your own eyes if the screws are loose, broekn or who knows which state.
let me know.

pencoj1, 2019-04-02:
tightened the screws inside with Julien. Solved

closed 2019-04-03 by julijenv