robotology/mesh-support #732: Left lower arm board restarts

hu-yue, 2018-10-16:
### Description of the failure

We lost control of the left lower arm joints (from wrist prosupination), we saw that the board was restarting. After restarting it was fine.

### Detailed conditions and logs of the failure

We ran the `WalkingModule` and it was not able to acquire joint limits from the boards. Therefore we tried to move the wrist prosupination using the motorgui and it did not move, even if it was shown as being in `position` control mode.

Unfortunately due to the necessity of reducing network communications, we don't have any log.

Gif of the reset of the board:
[image]

hu-yue, 2018-10-17:
@julijenv it happened after some time, we were trying to launch the walking and retargeting. We could launch them at the beginning, but we did move, then we closed the modules and then they were missing. We first lost the connection to the left hand fingers, then we lost the right hand fingers, then the wrist. We did suspect it could be a connection issue, we will try the same network setup tomorrow to check this.

julijenv, 2018-11-19:
has been resdtored

closed 2018-11-19 by julijenv