robotology/mesh-support #733: Wrist cables broken left+right wrist

Tobias-Fischer, 2018-10-22:
### Description of the failure
When shutting down `yarprobotinterface`, the robot suddenly started shaking. By the time we pressed the red button, the right shoulder was broken, as well as several cables of the left and right wrist. We managed to fix the shoulder, but require some remote assistance to fix the wrist.

### Detailed conditions and logs of the failure
Please see the attached pictures for more details.

Right arm top-view - see one end of the broken cable (going to the tensioner)
[image]

Right arm side view - see other end of the broken cable (sticking out on top)
[image]

Left arm view - similar damage as the right arm
[image]

I am not sure what the correct procedure is to replace those cables. It would be great if you could briefly describe what to do here, or if you're available on Skype to have a quick chat. I think we will be okay once we know the rough procedure, but we don't want to take the wrong things apart.

If you need any more pictures from different angles, please let us know.

Best and thanks a lot in advance,
Tobi

/cc
@umgnunes

umgnunes, 2018-10-26:
Hi.

We have managed to fix both wrists.

[image]
[image]

We have ensured that the cables were at least as tense as the other ones that were not broken. If more pictures are needed for you to corroborate, please let us know.

Now, we still have a broken cable related to the left arm middle finger, which again we are not sure how to correctly proceed to replace it. It would be great if you could briefly describe what to do here, or if you're available on Skype to have a quick chat.

Thank you for the support,
Urbano Miguel Nunes

/cc
@Tobias-Fischer

julijenv, 2018-10-26:
Hi @Tobias-Fischer , @umgnunes ,
great job :+1: 
did you also succeed to redo the proxximal joint , I remember it was broken too?

Tobias-Fischer, 2018-11-21:
Hi @julijenv,
We got the cable for the proximal middle finger joint and are in the process of repairing. We are feeding the wire through the tube, but it gets stuck at some point. Do you have any advice on what to do? Would be great to have a quick Skype chat.

Best, Tobi and Urbano

julijenv, 2019-01-18:
it has been done from what I recall

closed 2019-01-18 by julijenv