robotology/mesh-support #735: Torso, head and battery pack cover broken

hu-yue, 2018-10-25:
### Description of the failure

The torso cover broke due to the arm pushing towards the torso. It was glued by @julijenv and now broke again, we might just glue it again.

The head cover and the battery pack cover broke during falling on the back. Furthermore, we digged four holes on the head cover to accommodate the Ericsson device, so it should be changed. The battery pack cover is not broken yet, but it has two cracks, if the robot falls another time on the back, it might just break.

DanielePucci, 2019-01-25:
Hi @julijenv any update on this?

DanielePucci, 2019-02-27:
Hi @julijenv, do you have any update on this?

julijenv, 2019-03-14:
Hi @DanielePucci , it has been mounted the the T shape of the heaad cage as well as reshaped the mettlaic part hold ing the inertial sensor Bosch NB0055 (something like that). The torso was reglued and Im still waiting to get the piece to mount it,.

closed 2019-10-12 by pattacini