robotology/mesh-support #737: USB cable in head (?) probably

kt10aan, 2018-10-31:
### Description of the failure
The PC104 became unreachable. We tried to boot the USB on a laptop and it works perfectly. What made us thin about cabling issues was the the blue light of the stick doesn't come up not even once when the CPU us turned on.

### Detailed conditions and logs of the failure
We checked the cable for continuity. We found the following problem, one line did not  in one of the connection between the USB connector of the USB stick and the connector on the board. 
The first ones from the left have no continuity. 
I post it here in case you think there maybe also be something else involved that I have to check.

[image]

kt10aan, 2018-10-31:
I want to add that the issue with pin 1 is also with the other USB connector that goes to the LEDs of the face, so I assume that the bad connection is at the board connector and not at the USB ones.

julijenv, 2018-10-31:
Hi @kt10aan ,
I dont really understand how you could assume that the problem is coming from the board and not from the wires if you dont have contiunuity in between pin 1 of the connector atttached to the pc104 and the pin 1 of the usb connector, I will assume that the problem is the wires not the board.
Could you double check that you dont have continuity at the level of those wires for both usb?
If no continuity please try to fix at least one of them After restoring the 2 usb and connector (set of connectors and wires from your pictures) see if the usb stick got unlighted when plugged in.
If  no light, Could you try to check directly on the pc104 if u have 5V in between pin 1 and 3 pin more on the right side (watch out not touching,2 pins on the same time) this will be you ultimate test to declare that the 5V pins are defective on board pc104.
let us know

kt10aan, 2018-10-31:
OK, my bad, misunderstanding, when I said board-side, I meant the female connector that goes to the board, not the actual board itself! I opened up the connectors and I will make a new one tomorrow. Indeed, the brown cable was reattached. Just for reference, could you write me the colours of the cables that go to each pin?

julijenv, 2018-10-31:
Hi @kt10aan ,
on the last version of this cable the colour has been changed but if you take a look at the old one you could easily identify where goes who.
[image]
hope this will be of any help

kt10aan, 2018-11-05:
Ok here is how the situation is:

I fixed the cable and if we take as USB1 the USB stick female connector with pins 4-1, from left to write
and USB2 the other USB connector, again with pins 4-1 from right to left.
And if we take the rectangular connector as

1 | 2 | 3 | 4
5 | 6| 7| 
from left to right
then the following connections have the following continuity:

USB1 Pin 4 --> 5
USB1 Pin 3 -- > 2
USB1 Pin2 --> 3
USB1 Pin1 --> 4
And

USB2 Pin 4 --> 1
USB2 Pin 3 --> 6
USB2 Pin2 --> 7
USB2 Pin1 --> 4

So all seem ok, connection wise  but still I get no power on the USB (which functions fine on other machines).

1) Just for the sake of checking everything, could you send me a photo of where the connector goes into the pc104?

2) Could you also point in this photo where are the pins 1-3 on the board that i can check for the existence of 5V?

kt10aan, 2018-11-05:
Also to mention the the green LED  in the bottom lefft corner of the PC104 is lit.

julijenv, 2018-11-05:
hi @kt10aan ,
hope this will help you,
[image]
cheers

kt10aan, 2018-11-05:
I think in our iCub there is a gap of three double pins between this connector and the one at the left

julijenv, 2018-11-05:
the connector is actually right the most left on the longer connector I took a picutre without me holding the usb connector which could lead you to misunderstanding.
[image]
plug it directly to the left side of the longest header.
On PC104:
header 6x2¦headerr 6x2¦long header(on the left of it plug your connector)

kt10aan, 2018-11-06:
This is exactly where I plugging it. I double check the cable many many many times and all connections work ok. But stil no light on the USB stick (and it works fine in other computers).

Could we have a skype meeting today? ( we have 8 hours time difference so early would be preferrable).

kt10aan, 2018-11-07:
My mistake. The voltage between PINs 1 and 4 and pins 5 and 4 is 3V, on the board and on the usb connector. But not 5 V.

julijenv, 2018-11-07:
R u sure that the pc104 start. do the fanspin?
Maybe the usb led dont turn on but 3V may probably enough to power the memory part of the usb key, can you ping the robot? Im gonna chekc if there are other 5V pin somewhere on the pc104 pinouts.
Ill let you know

kt10aan, 2018-11-07:
The fans start. The green led in the bottom left corner is on. So I assume the pc104 starts ok. All other boards have lights on also. Also, when I turn on the CPU power switch, the servos twitch a little, which is normal.

The USB stick lights up wherever else I have put it. Just not on the PC104. But I cannot ping it. (I cleared the udev net rules after booting in on a laptop.)

I even tried with another USB key with a fresh install on it. No ping.

julijenv, 2018-11-07:
Is it the same USB key as before??? or is it a new USB key?!?!?

kt10aan, 2018-11-07:
I tried the old USB key, no light no ping. I tried *another* USB key with a fresh install on it and the result was the same.

julijenv, 2018-11-07:
Hi @kt10aan ,
please refer to this document [here](http://wiki.icub.org/images/d/dc/EMBEDDED_LOGIC_-_PB945-_CPU_board.pdf) to check the 5V if it is somewhere else and if you could change the USB port, it is written p.24 that all USB are 5V pin resettable fuse so theritically after a restart the %V pin should be working fine again. please refer p.25 and check if 5V pins are ok for pin 13 and 14 for USB3 and USB4 connector if yes, just move the connector (pins)
Just as a reminder the usb stick should be USB2.0 not USB3.0.

kt10aan, 2018-11-09:
Hi @julijenv ,

indeed, from USB3 and USB4 I could get something around 3.7 volts. Using the connector starting from pins 7 and 8, and putting it rotated, did the job. USB stick lights app and I can boot and ssh into the PC104 :)

However, at the same time something else happened. The same thing that had happened when you were here. As soon as I turn on the motors, both power supplies go to off. I haven't touch anything on the power supplies since you fixed them.

Do you remember what the solution was?

kt10aan, 2018-11-09:
You had put a capacitor in our to power supply. In the meantime, we did get the STPS15L25D diodes and the connectors you had told me.

Should I remove the capacitor and add the diode? 
[image]

kt10aan, 2018-11-09:
I "fixed" the power supply issue by putting both in Local mode and manually dialling in the settings.
But it would be nice to have it working as Remote again. In any case, the main issue is solved so you can close the issue. Thanks :)

julijenv, 2018-11-09:
Hi @kt10aan , could uou please try to cactch me on skype and try to see if ever there isnt another htings that could prevent the pc104 to work properly and caonsequently make the 2 power supply to tuen off.
From what I remember I just increase the OverVoltage on the power supply of the pc104 i think maybe both i dont remember exactly. Let's  see that together if you want.
let me knwo

kt10aan, 2018-11-09:
You had increased the current on the top power supply to 55A. In any case, the robot works as it is, with USB3 and USB4, and both power supplies in Local mode, which is good because we are in the middle of experiments :)

I will try to catch you on Skype on Monday.

julijenv, 2018-11-09:
ok sounds good to me