robotology/mesh-support #738: Segmentation Fault - IGazeControl->get3DPoint()

RainAlexandra, 2018-11-11:
I'm new to the iCub Gaze Interface and I'm trying to get my robot to ook at a point. To do this I do:

```
Property options;
options.put("device", "gazecontrollerclient");
options.put("remote","/iKinGazeCtrl");
options.put("local","/client/gaze");
PolyDriver clientGazeCtrl(options);

IGazeControl *igaze=NULL;

if (clientGazeCtrl.isValid()) {
   clientGazeCtrl.view(igaze);
}

int camSel = 0;
Vector px(2);
px[0] = 100;
px[1] = 40;
double z = 1.0;

Vector fixpoint(3);
printf("retrieving 3D point\n");
igaze->get3DPoint(camSel,px,z,fixpoint);
igaze->lookAtFixationPoint(fixpoint);
igaze->setTrackingMode(true);
igaze->waitMotionDone();

clientGazeCtrl.close();
```

Here is the result I'm getting.
```
yarp: Port /imageProc/target/in active at tcp://100.88.4.22:10083/
yarp: Receiving input from /imageProc/target/out to /imageProc/target/in using tcp
Cannot load plugin from shared library (gazecontrollerclient)
(gazecontrollerclient: cannot open shared object file: No such file or directory)
yarpdev: ***ERROR*** could not find device <gazecontrollerclient>
retrieving 3D point
Segmentation fault (core dumped)
```

Could it have something to do with the way I am initializing my vectors?

julijenv, 2018-11-19:
Hi @RainAlexandra ,
which robot are we talking about?!?!

julijenv, 2018-11-19:
@pattacini , could you please take a look at this issue or give it to who could fit to answer, thx in advance