robotology/mesh-support #739: Head cage and cover broken

S-Dafarra, 2018-11-11:
### Description of the failure
As @julijenv noticed, the greeny's head cage is bent, while the plastic hear ring holding the face cover is broken. This is probably due to some fall the robot has undergone. 

### Detailed conditions and logs of the failure
- Face cage bent
[image]
- Plastic ear ring broken
[image]

cc 
@DanielePucci @nunoguedelha  @gabrielenava @GiulioRomualdi

DanielePucci, 2019-01-08:
@julijenv 
We would need support on this. May you give us a hand?

julijenv, 2019-01-09:
Hi @DanielePucci ,
We saw with M. SAvoldi this morning that the frame is bent has from the picture already taken above (RC_IIT_030_P_010) but we also found out that the T shape holding the tilt motor and making a cube with the frame is also bent but more badly(RC_IIT_010_P_051).
I will see if we have this latter piece on stock and try to replace it, rebending the first piece as it is simpler and a quick fix for the overall head.
The best fix would be to make n upgrade but I fear that the robot will fall again so to me , quick fix could be less expensive and take less time on the robot.

DanielePucci, 2019-01-09:
@julijenv 
Thanks, great!

DanielePucci, 2019-02-27:
Hi @julijenv, do you have any update on this?

julijenv, 2019-03-14:
Hi @DanielePucci ,
cover head had been recreated 2/3 weeks ago, 2 days ago with the help of M. SAvoldi we replaced the T-shape of the face cage and we also the reshape the plate on which lie the inertial sensor

closed 2019-03-14 by julijenv