robotology/mesh-support #740: Wrong Value of MotioncontrolVersion in yarprobotinterface

stu105502, 2018-11-20:
### Description of the failure

Hello,

Currently, I'm updating our iCub on the newest versions. I assume that everything was installed properly, but there seems to be a general Problem of our robot-configuration files. The installed embObjMotionControll driver seems to expect version 5, but receives version 3. This is a problem we seem to have since last year.

I have seen your templates in the github, but I really don't see, which parts has to be adjust and I didn't found a log that explains, where are the precise changes in each version

### Detailed conditions and logs of the failure

```
icub@icub-head:/usr/local/src/robot/old_versions/Nov18/robots-configuration/build$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini
||| finding file [config]
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]Preprocessor complete in:  0.0150881 s 
||| finding paths [plugins]
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/plugins] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/old_versions/Nov18/yarp/build/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/yarp/build/share/yarp/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/icub-main/build/share/iCub/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/icub-main/build/share/iCub/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /icub/yarprobotinterface active at tcp://10.0.0.2:10002/
[INFO]startup phase starting... 
||| finding paths [plugins]
[DEBUG]eth::parser::print(pc104Data) for PC104: 
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345 
[DEBUG]PC104/PC104TXrate =  1 
[DEBUG]PC104/PC104RXrate =  5 
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms 
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms 
[WARNING]in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
[DEBUG]eth::parser::print(boardData) for BOARD head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES: 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.20 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS: 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60 
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.20 (head-eb20-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec 
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.20 
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ------ ATTENTION!!!! Wrong value of <MotioncontrolVersion> parameter !!!! --------------------------------------------------------------------------------------- 
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ------ This means that the configuration files of this device are not compatible with my parser, so I cannot start.  
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ------ I need version  5 , but in configuration files have version  3 . 
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ------ Please update configuration files in robots-configuration repository. (see http://wiki.icub.org/wiki/Robot_configuration for more information).  
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ------ If the problem persists contact icub-support@iit.it . 
[ERROR]embObjMC BOARD head-eb20-j0_1 (IP 10.0.1.20)  ---------------------------------------------------------------------------------------------------------------------------------------------------------------- 
[ERROR]BOARD head-eb20-j0_1 (IP 10.0.1.20)  Missing motion control parameters in config file 
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open
[... 42 lines cut ...]
[ERROR]park device do not exists 
[ERROR]Cannot run park action on device head-calibartor 
[INFO]All actions for action level 1 of interrupt1 phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[WARNING]There was some problem running actions for interrupt1 phase . Please check the log and your configuration 
[INFO]interrupt1 phase finished. 
[ERROR]Error in interrupt1 phase... see previous messages for more info 
[INFO]shutdown phase starting... 
[INFO]Entering action level 5 of phase shutdown 
[ERROR]head-mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device head-mc_wrapper 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[WARNING]There was some problem running actions for shutdown phase . Please check the log and your configuration 
[INFO]shutdown phase finished. 
[ERROR]Error in shutdown phase... see previous messages for more info 
[INFO]RFModule failed to open.
yarp: 
 Warning an issue has been found, please update the code.
     Clock is not initialized: This means YARP framework has not been properly initialized. 
     The clock can be initialized with one of the following methods:
     - Create yarp::os::Network object or call yarp::os::Network::init()
     - Call useSystemClock()
     otherwise use yarp::os::SystemClock::nowSystem() and yarp::os::SystemClock::delaySystem() instead of Time::now() and Time::delay()
```

julijenv, 2018-11-27:
Hi @stu105502 ,
sorry for the late reply, I started the reply some days ago, and I thought I had sent it but indeed NOT :S.
So to reconfigure your files and relaunch correctly your head, you just need to edit the overall files
*-mec.xml inside iCubHamburg01/hardawre/mechanical/ and change the control version to 5 instead of 3 previously.
and then please download the 4 files from this [link](https://drive.google.com/drive/folders/1nyYFOFSr9NvECwWmN4FSAJ6Rd-3a7Z92?usp=sharing) and put them in the repository iCubHamburg01/hardware/motorcontrol. you may need first to backup your previous files appending .bak for example.
let me know if it works

stu105502, 2018-11-28:
### Description of the failure

Thank you very much! This helped a lot, but now there seems to be a problem regarding the head-calibrator. He doesn't create this yarp device correctly or at least it is told that the joint with the pid 0 can not be received. Maybe it is also important to mention that we updated our firmware just recently.

Do I have to recompile something or does this problem occur in the robot-config calibrator- and wrapper-file?

### Logs and so on

```
icub@icub-head:/usr/local/src/robot/old_versions/Nov18/robots-configuration/build$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini
||| finding file [config]
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]Preprocessor complete in:  0.0110092 s 
||| finding paths [plugins]
||| checking [/usr/local/src/robot/old_versions/Nov18/robots-configuration/build/plugins] (pwd)
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/old_versions/Nov18/yarp/build/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/yarp/build/share/yarp/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/icub-main/build/share/iCub/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/icub-main/build/share/iCub/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/old_versions/Nov18/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/old_versions/Nov18/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /icub/yarprobotinterface active at tcp://10.0.0.2:10002/
[INFO]startup phase starting... 
||| finding paths [plugins]
[DEBUG]eth::parser::print(pc104Data) for PC104: 
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345 
[DEBUG]PC104/PC104TXrate =  1 
[DEBUG]PC104/PC104RXrate =  5 
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms 
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms 
[WARNING]in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
[DEBUG]eth::parser::print(boardData) for BOARD head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES: 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.20 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS: 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60 
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.20 (head-eb20-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec 
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.20 
[WARNING]embObjMC BOARD  head-eb20-j0_1 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails) 
[INFO] from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 5032s 472m 633u: (code 0x00000034, par16 0x0001 par64 0x0000000000000014) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::verifyBoardPresence() found BOARD head-eb20-j0_1 with IP 10.0.1.20 after 0.00386429 seconds 
[DEBUG]EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb20-j0_1 with IP 10.0.1.20 
[DEBUG]EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb20-j0_1 with IP 10.0.1.20 : config mode + cleared all its regulars 
[... cut ...]
[WARNING]HeadV3_Calibrator : Calling park without calibration... skipping 
[DEBUG]head-calibartor finished park device head-mc_wrapper 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[INFO]interrupt1 phase finished. 
[INFO]shutdown phase starting... 
[INFO]Entering action level 5 of phase shutdown 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[INFO]shutdown phase finished. 
[INFO]RFModule failed to open.
```

stu105502, 2018-11-28:
Hi julijenv, 

thank you for your support! I have double checked every repository and I simply did not find any head_calibrator flag, but I have found two aspects that I can't follow as given in the instructions (http://wiki.icub.org/wiki/Compilation_on_the_pc104):

I was able to install in yarp enable properly "ENABLE_yarpmod_serialport" and "ENABLE_yarpmod_portaudio" as flags, but I the flag "ENABLE_yarpmod_serial" is missing completely. I assumed that this flag is deprecated.

Furthermore I'm not able to activate "ENABLE_icubmod_cfw2can", because the cfw2can api is not found. Could it be possible that we need a new update of the whole pc104? Our current version seems to be "iCub OS version 7.0-20160902"

julijenv, 2018-11-29:
Hi @stu105502 ,
can you check manually if you find the repository of the CFW2CAN_DIR and of course the file libcfw002.so as on the picture below.
[image]

> ENABLE_yarpmod_serial
I can confirm you that it has been deprecated.

To avoid making you wait and losing time, please contact me on skype: my ID is icub.support in order to clear out this issue ASAP to let you work faster, thx in advacne

stu105502, 2018-12-03:
Thank you very much than I will contact you per skype. I'm very thankful for your support!

julijenv, 2018-12-06:
Hi @stu105502 ,
After re-reading your last log it looks like that you may not have the cfw-002 board, this is why you cant access it and that you dont have the cfw2CANAPI_DIR and all the system variables related to it. To me it akes sense becasue the .bashrc_iCub dont mention cfw2 anywhere but you can still use the other repository and make the robot work. So double check this you could just ccmake your old reposistories (yarp and icub-main) and check the overall list of option and check if there are the same as the one dfrom the new repositories. If you cant see listed cfw2 (or the flag is OFF) please do the same on the new repositories, and try again to compile and launch robotinterface.
let me knwo

stu105502, 2018-12-20:
Hi julijenv,

we have now run a full updated live-version, recompiled everything and we receive the same error message. The head_calibrator still does not find the joint with the PID 1. Furthermore, we actually seem to have eth-boards. At least, this is told by the FirmwareUpdater and even with the installed cfw2can api our error remains the same.

Could you guys maybe take a look on the robot-configuration files. If everything is really as it should?

julijenv, 2018-12-21:
Hi @stu105502 , can you please send me the new log file?!?!? because logically the error message about version 3 instead of 5 should have been removed, by the 4 files link previuuiously.
Your robot is ethernet base, you shouldnt have the cfw002 module to be enabled so.
You shouldnt have any error related to it.

stu105502, 2019-01-07:
Hello @julijenv and a happy new year,
I have added the new error log, but it seems to be exactly the same error. Out of some reasons, we are able to satrt the head-calibrator as parametricCalibratorEth, but two lines later we receive the error message that the head-calibrator does not exist. Somehow, our  system is not able to detect the head-calibrator in its network. I have had run the yarpserver at the server laptop and also on the pc104 to see if this is a network problem between server and laptop, but in both cases occur the same error.

```
icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/./icub_all.xml 
[DEBUG]Preprocessor complete in:  0.00927162 s 
||| finding paths [plugins]
||| checking [/home/icub/plugins] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubHamburg01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /icub/yarprobotinterface active at tcp://10.0.0.2:10002/
[INFO]startup phase starting... 
||| finding paths [plugins]
[DEBUG]eth::parser::print(pc104Data) for PC104: 
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345 
[DEBUG]PC104/PC104TXrate =  1 
[DEBUG]PC104/PC104RXrate =  5 
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms 
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms 
[WARNING]in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
[DEBUG]eth::parser::print(boardData) for BOARD head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES: 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.20 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768 
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS: 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb20-j0_1 
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE 
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60 
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.20 (head-eb20-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec 
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.20 
[WARNING]embObjMC BOARD  head-eb20-j0_1 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails) 
[INFO] from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 7342s 636m 892u: (code 0x00000034, par16 0x0001 par64 0x0000000000000014) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::verifyBoardPresence() found BOARD head-eb20-j0_1 with IP 10.0.1.20 after 0.00391603 seconds 
[DEBUG]EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb20-j0_1 with IP 10.0.1.20 
[DEBUG]EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb20-j0_1 with IP 10.0.1.20 : config mode + cleared all its regulars 
[DEBUG]EthResource::setTimingOfRunningCycle() for BOARD head-eb20-j0_1 with IP 10.0.1.20 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle 
[INFO]EthResource::askBoardVersion() found BOARD head-eb20-j0_1 @ IP 10.0.1.20 of type mc4plus with FW = ver 3.17 built on 2018 Jul 30 15:51 
[... cut ...]
[WARNING]HeadV3_Calibrator : Calling park without calibration... skipping 
[DEBUG]head-calibartor finished park device head-mc_wrapper 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[INFO]interrupt1 phase finished. 
[INFO]shutdown phase starting... 
[INFO]Entering action level 5 of phase shutdown 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[INFO]shutdown phase finished. 
[INFO]RFModule failed to open.
```

stu105502, 2019-01-07:
Furthermore, here are the information that our server provides, when we try to start yarprobotinterface:

```
 * register "/icub/yarprobotinterface"
   + set "/icub/yarprobotinterface" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/yarprobotinterface" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/yarprobotinterface" ips "127.0.0.1" "10.0.0.2" "10.0.1.104" "::1" "fe80::213:95ff:fe20:1a36%2" "fe80::20c:8bff:fe24:762%3"
   + set "/icub/yarprobotinterface" process 11434
-> announce "/icub/yarprobotinterface"
 * register "/icub/head/rpc:i"
   + set "/icub/head/rpc:i" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/rpc:i" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/rpc:i" ips "127.0.0.1" "10.0.0.2" "10.0.1.104" "::1" "fe80::213:95ff:fe20:1a36%2" "fe80::20c:8bff:fe24:762%3"
   + set "/icub/head/rpc:i" process 11434
-> announce "/icub/head/rpc:i"
 * register "/icub/head/command:i"
   + set "/icub/head/command:i" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/command:i" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/command:i" ips "127.0.0.1" "10.0.0.2" "10.0.1.104" "::1" "fe80::213:95ff:fe20:1a36%2" "fe80::20c:8bff:fe24:762%3"
   + set "/icub/head/command:i" process 11434
-> announce "/icub/head/command:i"
 * register "/icub/head/state:o" "..." "..." -1 "yarp/vector"
   + set "/icub/head/state:o" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/state:o" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/state:o" ips "127.0.0.1" "10.0.0.2" "10.0.1.104" "::1" "fe80::213:95ff:fe20:1a36%2" "fe80::20c:8bff:fe24:762%3"
   + set "/icub/head/state:o" process 11434
-> announce "/icub/head/state:o"
 * register "/icub/head/stateExt:o"
   + set "/icub/head/stateExt:o" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/stateExt:o" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer portmonitor priority tcpros rossrv shmem xmlrpc
   + set "/icub/head/stateExt:o" ips "127.0.0.1" "10.0.0.2" "10.0.1.104" "::1" "fe80::213:95ff:fe20:1a36%2" "fe80::20c:8bff:fe24:762%3"
   + set "/icub/head/stateExt:o" process 11434
-> announce "/icub/head/stateExt:o"
 * unregister "/icub/head/rpc:i"
 * unregister "/icub/head/command:i"
 * unregister "/icub/head/state:o"
 * unregister "/icub/head/stateExt:o"
 * unregister "/icub/yarprobotinterface"
```

stu105502, 2019-01-07:
sry to bother you @julijenv,
I found the error in the whole procedure. In our robot-configuration files in calibrators->head_calib.xml seems to be a small spell-mistake. In line 5 you have declared the parametricCalibratorEth as "head-calibator", which causes big trouble for the attachment process of the head_mc_wrapper.
Now everything works perfectly fine, but I would appreciate it, if you could update our robot-configuration-files by correcting the spell mistake and change the four files in our hardware->motorControl folder with the files you have already sent me.

Thank you very much for your support and your time. I appreciate your help very much!

Regards,

Matthias

closed 2019-01-07 by stu105502

julijenv, 2019-01-07:
Hi @stu105502 ,
I, myself, make sure that the files are uptodate and have been pushed and commit correctly, double check if you ish but it should be alright now.
cheers, thx for your debugging