robotology/mesh-support #743: High current load melted wires in waist

Iaxama, 2018-12-06:
### Description of the failure
While launching yarprobotinterface high current load in a wire at the level of waist caused melting of surrounding plastic and of the board attached to it. 
During the startup phase of yarprobotinterface the calibration never started, however the robot slightly moved the torso before starting to smoke, forcing us to shut it down.

### Detailed conditions and logs of the failure
I attach pictures showing the damages and the log of yarprobotinterface.

[image]
[image]
[image]
[image]
[log.05dec18.h1245.txt](https://github.com/robotology/icub-support/files/2652415/log.05dec18.h1245.txt)

MrAndrea, 2018-12-17:
Thanks to @julijenv and Emiliano who replaced the cables and boards. I programmed the BDC board inside the backpack since it was not already programmed, and I decided to put the hardware limit of the maximum current to 32A by replacing the shunt resistor on the board with two 3mohm resistors in parallel.

julijenv, 2018-12-17:
#748

closed 2018-12-20 by julijenv