robotology/mesh-support #745: Right leg hip pitch in hardware fold, enconder invalid measure

fjandrad, 2018-12-10:
### Description of the failure
After performing some movements on the pole (usual movements), the right leg hip pitch went into hardware fault and it can not be idled.

### Detailed conditions and logs of the failure 
When rising the robot with the crane the right leg hip pitch went into hardware fault 
 <ERROR>  from BOARD 10.0.1.8 (right_leg-eb8-j0_3), src LOCAL, adr 0, time 1143s 344m 163u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
[log_icub-head_yarprobotinterface_8327.txt](https://github.com/robotology/icub-support/files/2663603/log_icub-head_yarprobotinterface_8327.txt)

cc @GiulioRomualdi @DanielePucci @S-Dafarra @gabrielenava

fjandrad, 2018-12-10:
Update: 
After restarting the motors it did not happen again. We might try a few more times tomorrow and see if it repeats.

GiulioRomualdi, 2018-12-11:
This morning Emiliano has checked the robot. It seems there are no problems related to the connection and since the problem disappears after restarting the robot, we cannot exclude that the issue is related to the board itself.

fjandrad, 2018-12-12:
Update: 
It happened again once more. Rebooting the motors makes the error go away.

I do not know if it might be caused by static discharges (everyone is having) .

After rebooting and trying to discharge before touching the robot , the error did not present again.

closed 2019-01-18 by julijenv