robotology/mesh-support #746: iCubLondon01 - short-circuit on motor board

umgnunes, 2018-12-11:
Hi,
We are having problems with one of the boards of the torso. It might be related to  #663, because almost everything seems to match.

### Description of the failure

After our adventure fixing the wrists and middle finger cables, we proceeded to calibrate iCub. Everything was going well, until one time when we were shutting iCub down (I was about to stop robotInterface to be precise, because of an old problem related to #216 ) and suddenly iCub turned off. We suspected immediately it was a short-circuit and we found out that indeed there is a short-circuit on the BLP_001_B 0BPB4 electronic board, between the pins VCC and GND.

[image]

Thus, we would like to order another one.
Something must not be okay and we would appreciate if you could give us some insight on possible issues.
Thanks.
@umgnunes and @Tobias-Fischer

julijenv, 2018-12-12:
Hi guys, @umgnunes @Tobias-Fischer ,
I'm quite puzzled by hearing that the same board keep burning and I do believe that something else make this happening, I guess you have some short-circuit somewhere on the line or that some wires are pinched or worn and make contact were they should not. In the other hand it could also be due to the control boad BLL which could not have really well appreciated the 2 short circuit of the power boards.
Please contact Andrea Derito to get a quotation for the BLP.
Please check if you dont have awy cables worn somewhere on the bottom part but also maybe by looking on the back connector of the robot, view from below.
let me know

umgnunes, 2018-12-13:
Hi @julijenv ,

We have already ordered the BLP and the associated BLL.
I have taken a look at the back and I did not find anything out of order (no pinched or worn cables).

[image]

Aside from some out of place cables (inside blue circle), they seem to be alright (inside red circle; these are the cables associated with the short-circuited BLP board). Not sure how to proceed further.

Thanks,
@umgnunes

julijenv, 2019-03-18:
@umgnunes , @Tobias-Fischer ,
any news?

Tobias-Fischer, 2019-03-18:
All done thanks to @umgnunes. Thanks for checking, Julien!

closed 2019-03-18 by Tobias-Fischer

umgnunes, 2019-03-18:
Hey Julien.

Everything is done and so far iCub is working, although the recurring blue light problem keeps appearing if iCub works for a somewhat long period. Also, the back keeps getting quite hot (not to the point of burning though). Is it possible to lower the input voltage to say about 50V (we are applying 52V due to iKart)?
Thanks!