robotology/mesh-support #747: Right shoulder cable broken

GiulioRomualdi, 2018-12-13:
### Description of the failure
While performing the Yoga ++ demo, the right shoulder got broken. Spikes coming from the encoders maybe cause the issue.

[image]
 
cc @S-Dafarra @DanielePucci @julijenv

kouroshD, 2018-12-13:
From the following figure you can see the spikes coming from the left arm encoders

[image]

In the following you can find the data collected
[backupYogaAfterSpikesLeftArm.zip](https://github.com/robotology/icub-support/files/2676888/backupYogaAfterSpikesLeftArm.zip)

julijenv, 2018-12-13:
Hi @kouroshD , could you please tell me which joint are we talking about??? which encoder (AEA I presume)?
for @GiulioRomualdi , it has been replaced 2 cables of the left shoulder
IITCODE Qta Description
4765.2 x 1 : cable shoulder 8
4758.1 X 1 :shoulder cable N1

GiulioRomualdi, 2018-12-13:
@julijenv thank you for your prompt intervention. 
As regards the left arm encoder I was talking about the shoulder and elbow encoders. Here a screenshot af the data collected  using simulink. 
* left_arm 1: shoulder pitch
* left_arm 2: shoulder roll
* left_arm 3: shoulder yaw
* left_arm 4: elbow

[image]

The issue seems to be similar to that one https://github.com/robotology/icub-support/issues/624

traversaro, 2018-12-17:
The fact that the spikes are synchronized among the different axis may suggest that the spike does not come from the raw encoder, but rather from some bug in the software architecture. It is easy to reproduce the issue? It would be great to get discrete datasets (no plots) and also read the motor encoders to understand what the problem is.

DanielePucci, 2019-01-06:
@kouroshD - helped by @GiulioRomualdi - please run the Yoga++ demo and see if these spikes occur again

GiulioRomualdi, 2019-01-08:
Today we tested the Yoga++ demo several time (left and right). We did not notice any kind of spikes. I think we can close this issue

closed 2019-01-08 by GiulioRomualdi

DanielePucci, 2019-01-09:
That's unfortunate, but clearly if we are not able to reproduce the issue, closing is the right move to do. Thanks @GiulioRomualdi and @kouroshD