robotology/mesh-support #748: iCubGenova02 reconfiguration and recalibration

arrenglover, 2018-12-14:
### Description of the failure

The iCubGenova02 Purple robot was reinstalled from a blank image and needs to be completely reconfigured and re-calibrated to reach a working state again. Now demos have been finished I would like to do this in as thorough manner as possible. 

### Detailed conditions and logs of the failure

The first step is to make sure all the correct procedures are performed - I am not sure who has the most experience with all the steps required to go from a blank robot to a completely configured robot. Please edit this description to add anything else needed.

- [ ] check this list of things to do is not missing a crucial step

- [ ] check the firmware current limits on the motors have been reset to normal levels after the jumping experiments to ensure damage will not occur again.
- [ ] check the yarprobotinterface can start without errors, and all encoders can be correctly read.
- [ ] check raw encoder calibrations and limits are correct.
- [ ] check the yarprobotinterface can start while performing the calibration start-up.
- [ ] perform a fine-tuning of the home position of the robot.
- [ ] check PID values are correct
- [ ] perform a calibration of the icubeyes - both leopard and ATIS

arrenglover, 2018-12-17:
@Iaxama @julijenv - what was the status for the firmware current limits? Should they have been reset to normal already, or is there something we need to do still?

julijenv, 2018-12-17:
Hi @arrenglover ,
I have no idea about his feature, you may need to refer to @MrAndrea about it, he will know for sure.

ale-git, 2018-12-17:
I'm going to reload the original firmware on the legs.

arrenglover, 2018-12-17:
thanks @ale-git !

julijenv, 2018-12-17:
#743

ale-git, 2018-12-17:
I've updated the firmware in the legs but there's still an issue about the torso 2FOC board: the replaced board is in address conflict with its mate on the same can bus. The address must be changed but I have to disconnect one of the two connectors and I need the production support for that. PLEASE DO NOT USE THE TORSO UNTIL IT'S FIXED!

MrAndrea, 2018-12-17:
As I described in the #743, the hardware current limit on the high voltage supply now is about 32A.

ale-git, 2018-12-17:
I've reloaded the original firmware on 2FOC boards, the robot is usable.

arrenglover, 2018-12-18:
I performed a test today and managed to get the event-driven demo running. The only issue that I can notice immediately is that the head yaw has a very noisy motor. Thanks @MrAndrea and @ale-git

arrenglover, 2019-02-06:
@pattacini - we are having some troubles with the final steps of calibrating the robot - in particular the estimate of the position of the target in the robot reference frame from the visual position, i.e. so this depends on both the kinematic chain and visual calibration. Do you know if we should be using the head version "V3" since we were upgraded?

julijenv, 2019-08-01:
#781

closed 2019-10-12 by pattacini