robotology/mesh-support #749: Left arm torque problem

gsaponaro, 2018-12-14:
### Description of the failure

When we set the left arm to zero torque mode, it is very hard to move the arm (it's quite stiff).

By comparison, the right arm does not have this issue: it is easy to move when zero torque-controlled.

In terms of software, we are using the xml files from [robots-configuration/iCubLisboa01](https://github.com/robotology/robots-configuration/tree/master/iCubLisboa01) (master branch).

We already re-flashed the firmware (strain.hex).

### Detailed conditions and logs of the failure

We are setting the arm to torque mode using `demoForceControl`, selecting "zero torque".

In `yarpmotorgui`, we observe good Nm readings, which do change correctly when we move the arm. Screenshots:

[image]

[image]

In `yarprobotinterface`, we observe the following warnings when we switch from position control to torque control:

```
[DEBUG]sharedcan [1] msg from board 1: CMODE SET:3 
[WARNING]setControlModeRaw delay ( j:0), current mode: pos, requested: torq
[DEBUG]sharedcan [1] msg from board 1: CMODE SET:3 
[WARNING]setControlModeRaw delay ( j:1), current mode: pos, requested: torq
[DEBUG]sharedcan [1] msg from board 2: CMODE SET:3 
[WARNING]setControlModeRaw delay ( j:2), current mode: pos, requested: torq
[DEBUG]sharedcan [1] msg from board 2: CMODE SET:3 
[WARNING]setControlModeRaw delay ( j:3), current mode: pos, requested: torq
[DEBUG]sharedcan [1] msg from board 3: CMODE SET:3 
[WARNING]setControlModeRaw delay ( j:4), current mode: pos, requested: torq
[DEBUG]sharedcan [1] thread ran 595 times, req.dT:10[ms], av.dT:10.07[ms] av.loopT :0.04[ms]
```

julijenv, 2018-12-14:
Hi @gsaponaro ,

1.  could you please try to print on yarpscope the values of the 2 FT sensors to see if they are similar - I mean that you dont have a huge difference in between both of them- or worse if one is broken, which I dont think because the yarprobotinterface would have yelled at you.
2.  could you try to open the PID window when in torque mode (demoForceControl acitvated) adn tell me if for the left and right arm the values are similar.
3.  Last but not the least, could you please tell me the strain numbers and get the calibration matrix of each one of them, like that I could double check if the matrix is still ok or has been corrupted.

thx in advacne have a nice wkd

gsaponaro, 2018-12-14:
1. After switching on the motors, launching only `yarprobotinterface`, I can see that **`/icub/left_arm/analog:o` contains all zeros** (I tried restarting a few times, this is consistent), whereas `/icub/right_arm/analog:o` contains meaningful values such as: 69.4359741210938 -191.592132568359 28.0503845214844 -0.4144287109375 -2.79132080078125 -1.89404296875.
**yarpscope** plots that I produced with [this xml](https://gist.github.com/gsaponaro/dddc9424cdeac412181e862fb283f19b) - note the different scales along y axis:

[image]

2. OK, after starting `wholeBodyDynamics` and `gravityCompensator`, I activate zero torque mode with `demoForceControl`, then I go to `yarpmotorgui` and I check the Torque PID information for all joints. For example, j1 torque PIDs of the left arm shows:

[image]

I compared the two arms, values are similar, apart from the sign difference in some gains. You can see the values that we currently have in the `TRQ_PIDS` section of the configuration files [hardware/motorControl/icub_left_arm.xml](https://github.com/robotology/robots-configuration/blob/master/iCubLisboa01/hardware/motorControl/icub_left_arm.xml) and [icub_right_arm.xml](https://github.com/robotology/robots-configuration/blob/master/iCubLisboa01/hardware/motorControl/icub_right_arm.xml).

3. Can you please tell me how I can retrieve the strain numbers and the calibration matrices? That is, from which files or ports or GUI buttons?

Thanks a lot.

julijenv, 2018-12-16:
Hi @gsaponaro ,
R u sure that the FT sensor responds correctly, this zero everywhere would suggest me that the FT is dead or not repsonding or the matrix is 0 for all the elements.

1. if you still have canLoader (didnt update recently) just launch it and select net 1 and 3 and compare the information - you should see FT sensor and some BLL boards with the same ID, screenshot the two nets in order for me to understand what's going on.
if it doesnt work anymore,
2. Please use FirmwareUpdater, select CAN, and scan
on net 1 and 3 please double check the boards present.
The ID of the FT sensors should be indicated.

julijenv, 2018-12-16:
Hi @gsaponaro , YOU DID A HUGE MISTAKE ERASING THE EEPROM, YOU REMOVE THE MATRIX AND EVERYTHING INSIDE THE FT SENSOR, this is why it doesnt work, you shouldnt havfe done that..
now you need to open the left arm cover and try to retrieve the S/N on the FT sensor itself and afterwards we will be able to send you the matrix of calibration if we still have it.

julijenv, 2018-12-17:
Hi @gsaponaro ,
dont worry! ASAP you will have discovered the S/N engraved onto the F/T sensor, we will look inside our repository to retrieve the calibration matrix of your sensor and afterwards we will reflash it.
Later on, I will asked you to perform again the demoForceControl and if nothing works again, you will one more time use yarpscope to check the values that are sending out of the sensor.
Let me know, you can join me on skype anytime

gsaponaro, 2018-12-17:
Hi @julijenv , here is the sensor:

[image]

The engraved number is **094**.

The printed number is **84711**.

gsaponaro, 2018-12-17:
Hi @julijenv , thank you for the calibration file which you sent us on Skype: [`matrix_SN094.txt`](https://gist.github.com/gsaponaro/8be89d1819656c40bfd31b6e73707eab).

```
$ md5sum matrix_SN094.txt 
ee30d8105d1893701d56766061e120db  matrix_SN094.txt
```

In the PC104, I tried to flash that file onto CAN 1, ID 13 in the following ways:

* with the `canLoader` GUI
* with the command-line version `canLoader --canDeviceType cfw2can --canDeviceNum 1 --boardId 13 --firmware matrix_SN094.txt`
* with the further options `--calibration` and `--firmwareANDeeprom`
* changing the file extension from `.txt` to `.dat`

...but I always got the following error:

```
[ERROR]Failed Checksum
[ERROR]fatal error during download: abort
```

I did not see any advanced/privileged `--admin` option in `canLoader`. I only saw such option in the newer `FirmwareUpdater`, which right now we cannot use due to pending issue https://github.com/robotology/icub-main/issues/562

Can you confirm that our file is not corrupted? I pasted our md5 checksum towards the top of this message. I was thinking of a possible issue with Unix vs Windows line ending characters...
Any further suggestion?

Related link:
http://wiki.icub.org/wiki/Firmware#Firmware_update_.28Force.2FTorque_Sensors.29

gsaponaro, 2018-12-18:
Hi @julijenv , thank you for the correct file: [calibrationDataSN094.txt](https://gist.github.com/gsaponaro/6c5d08ed11ffb547a59031acf34c818d)

I re-flashed it successfully with `canLoader`:
```
Calibration file loaded!
[INFO]Data has been saved in EEprom correctly
```

Now, I got non-zero and constant data in the left arm analog port:
```
$ yarp read ... /icub/left_arm/analog:o
yarp: Port /tmp/port/1 active at tcp://10.10.1.53:10008/
yarp: Receiving input from /icub/left_arm/analog:o to /tmp/port/1 using tcp
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
```

What is the next step? Is it safe to run `demoForceControl` in zero-torque mode?

Thanks.

julijenv, 2018-12-18:
hi @gsaponaro , please proceed and check if you have a similar behaviour on both arm withthe demoForceControl  app.

gsaponaro, 2018-12-18:
New `yarpscope` plots:

[image]

OK, I'll talk to you on Skype.

gsaponaro, 2018-12-18:
Here is the comparison between the two arms:

**left arm F/T** (the one where we have the problem, and where we already re-flashed calibration matrix and `strain.hex`)

[image]

[image]

```bash
$ yarp read ... /icub/left_arm/analog:o
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
-0.0 -1974.0 -2222.0 -38.0 -42.0 -25.0
```
which as you can see are 1) constant values, 2) integer instead of decimal.

**right arm F/T** (which is working fine - even though the firmware is slightly older, I suggest we do not touch it)

[image]

[image]

Good values here:
```bash
$ yarp read ... /icub/right_arm/analog:o
71.9719848632812 -192.243621826172 32.9511413574219 -0.387725830078125 -2.91192626953125 -1.87939453125
72.0734252929688 -192.732238769531 32.9511413574219 -0.391998291015625 -2.91424560546875 -1.880859375
71.8705444335938 -192.515075683594 32.8866577148438 -0.396270751953125 -2.90728759765625 -1.884521484375
```

julijenv, 2018-12-18:
Hi @gsaponaro ,
as seen together, it has to be noted that flashing the calibration matrix files didnt seems to go perfectly well, in the SN section of the calibration window we could not notice appeareda  symbol at the end of the numbers and in the same way the first value of the 6 beaneath it goes to zero. which is different from the filess .dat above. @marcoaccame  could you tell us what is going wrong in teh flashing?
thx in advance

gsaponaro, 2018-12-18:
Exactly, the difference is between the full scale values in the bottom of the file [calibrationDataSN094](https://gist.github.com/gsaponaro/6c5d08ed11ffb547a59031acf34c818d) 

[image]

and the full scale values in the bottom left corner of the calibration window after having re-flashed using the above file:

[image]

marcoaccame, 2018-12-28:
Yes @julijenv and @gsaponaro, the data in the eeprom of the strain @ 10.0.1.1:CAN1:13 is not what is meant to be.

Let me see what we can do. I will come back to post a way through in a couple of hours.

marcoaccame, 2018-12-28:
Hi, here are some thoughts.

things which are wrong at the strain level.

- the first value of fullscale is wrong: 0 instead of 1804.
- the serial number string seems not terminated because it shows SN094plusdirtysymbols
- in the gui the values of last two values of the calibration tare are wrong: 0 and 0 instead of 
  112 and 65454.
- the channel[4] = -32736, which means and adc value of 32, hence saturated low.

there is a problem of eeprom values and maybe of a defective channel #4 (maybe due to a wrong value of offset).

we could do the following.

- attempt to write into the eeprom of the strain the correct calibration values.
- after that to re-evaluate acquisitions.
  
  
task 1: check the eeprom once more.

a. ask the calib gui to save to file the calibration data of the strain. it may be that the 
   calib gui does not show values correctly. then compare its content that with the original 
   file calibrationDataSN094.txt
   how to do it:
   - use button `Save Calibration`.
   - look for the saved file (... i don't remember the name the old gui is saving to)
   - compare directly or send it to me.
   
task 2: write the eeprom once more.   

b. load the calibration file calibrationDataSN094.txt directly from the calib gui.
   how to do it:
   - use button `Load Calibration`.
   - check the values on the gui
   - to save to eeprom you must press `Save to eeprom`. otherwise if you switch off and on the data is lost.
   - retrieve the calibration file once more as in step a.
     
task 3: evaluate once more the values

c. check he value of channel 4. 
   - if it is still saturated we may try to change the offset manually and see if there are any changes in the value
   

moreover, ... i looked at strain version 3.0.1: it is quite old and it may not transmit correctly the saturation info.
```
$ git log --follow strain.hex
commit d0e5dea70336eed6d88eb51846c744beea7a9ad4
Author: marcoaccame <marco.accame@iit.it>
Date:   Wed Feb 28 16:27:39 2018 +0100

    moved all can boards under CAN folder

commit d26d3fc240c52fef1d577315096512c2119c9835
Author: Valentina Gaggero <valentina.gaggero@iit.it>
Date:   Fri Sep 2 12:02:41 2016 +0200

    mais: 3.1.3, strain: 3.0.6, skin:2.17.1

    All have been compiled in MPLAX 3.35 IDE, but with the same compiler (Mplab c30 ver 3.31)

commit db4ba898b36b410acdaf329b5f71accaab47f62d
Author: Valentina Gaggero <valentina.gaggero@iit.it>
Date:   Wed Jul 6 11:51:55 2016 +0200

    Strain ver 3.0.5. added check writing to eeprom

commit ffb670db5f694a8f4c6f00e79d64bd547f599ac9
Author: Valentina Gaggero <valentina.gaggero@iit.it>
Date:   Thu Nov 26 12:09:52 2015 +0100

    strain ver 3.0.4

    compiled with source code at sha 4289700b50b70199a9bfc922c99b27730f8d841f
    of icub-firmware repository

commit e4aad972d297de971c9d9fa5a58392d6e973f396
Author: Valentina Gaggero <valentina.gaggero@iit.it>
Date:   Tue Nov 17 12:46:18 2015 +0100

    Strain ver 3.0.3
       - build with source in icub-firmware with SHA 3916d4356d8dff0374ff7d6182574c73f9112888
       - resolve issue num 33: Enhancing saturation info

commit 2ec061dbaa2f14ae50fd6a1d3781d3c2bb43177f
Author: Valentina Gaggero <valentina.gaggero@iit.it>
Date:   Fri Dec 6 15:05:19 2013 +0000

    version mais and strain 3.0.1.
    synchronized with sensorReaderDspic30f4013 sn rev 27599
    "Now fw and project of mais and strain uses iCubCanProtocolLib files.
    Updated configuration bits with new macros. Compiler C30 v3.31. MPLABIDE 8.91"

    svn path=/trunk/iCub/firmware/build/; revision=27600

```

marcoaccame, 2018-12-28:
I will be available from jan 2nd to help.

gsaponaro, 2018-12-31:
Hi @marcoaccame, thanks for your insight. Status:

**task 1: check the eeprom once more**

I used the button `Save Calibration` from `canLoader` and I recovered the file that it saved, here it is: [calibrationDataSN094_gui](https://gist.github.com/gsaponaro/9c71f652d7d94c153a2d74318ff8c598)
As you can see, it confirms that some values are wrong at the STRAIN level, compared to the "correct" file [calibrationDataSN094](https://gist.github.com/gsaponaro/6c5d08ed11ffb547a59031acf34c818d)

**task 2: write the eeprom once more**

I loaded the "correct" calibration file directly from the calib gui as per your instructions. I saved to EEPROM. I retrieved the calibration file once more: still, I get the "bad" file with those few wrong values.

**task 3: evaluate once more the values**

I confirm that channel 4 is still saturated as before, here is a fresh screenshot:

[image]

**sidenotes**

* I managed to fix the serial number string not being terminated because it showed SN094plusdirtysymbols, by manually editing it in the calib gui

* about the STRAIN firmware version: we have the recent one 3.0.6 in this arm with the F/T problem. The older firmware 3.0.1, which I mentioned for comparison, is in the other arm where the F/T is working correctly

* it would be great to analyze this issue via Skype / TeamViewer. Also, do you think it would be crucial to address the other issue https://github.com/robotology/icub-main/issues/562 first, so we can switch to using `FirmwareUpdater`? In that case thanks in advance also about that issue.

_Happy new year!_

marcoaccame, 2019-01-02:
Yes, it is of most importance to be aligned to master and use the `FirmwareUpdater`. I have replied on that in the issue robotology/icub-main#562 you cite. 

After the `FirmwareUpdater` works, I would like you to call it with the `--admin` option and launch the calibration window of the strain. Just for comparison.

gsaponaro, 2019-01-03:
Having managed to make `FirmwareUpdater` work, here is what I obtain with the admin option in the calibration window of the STRAIN of the faulty arm:

[image]

[image]

I have clicked `Save Calibration File` and I have obtained again the "bad" file (with some wrong values at fullscale and calibration tare), identical to [calibrationDataSN094_gui](https://gist.github.com/gsaponaro/9c71f652d7d94c153a2d74318ff8c598) that I linked earlier. I verified with md5 checksum that the file produced by `canLoader` is identical to this new file produced by `FirmwareUpdater`.

What would be the next step? Thanks a lot. (For talking on Skype usually the afternoon is better for me.)

marcoaccame, 2019-01-10:
Hi, sorry for my late reply but I am at home sick and I could read it only today.

Also, sorry for typos: I am with my phone.

Well, we still can see the values of ft and Calib bias wrongly memorised in eeprom. That is serious.

But also we see channel 4 and 5 far away from the value 0 as they should be when the offset of the front end analog stage is correctly set. 

Test 1 on calibration bias.
Press the button reset calib bias. Then save to eeprom. Power off, then on and then restart the giu. Is is all zero? Let’s see.
Then press ‘set calib bias’. There should be value of bias that send the adc equal to zero. Does it happen? Write the six values on paper. Save to eeprom, power off, then on and launch the gui. At the values of bias equal to the ones written on paper?

Let’s begin with that.

marcoaccame, 2019-01-10:
A clarification: write  to paper the six values of bias .... not of adc.

gsaponaro, 2019-01-10:
No problem @marcoaccame, and thanks for the pointers. Here is the result of the test:

> Press the button reset calib bias. Then save to eeprom. Power off, then on and then restart the giu. Is is all zero?

Yes: having done the above, and after a restart of motors + GUI, the biases are zero as displayed here:

[image]

> Then press ‘set calib bias’. There should be value of bias that send the adc equal to zero. Does it happen? 

Yes, it does. Screenshot:

[image]

The six values of bias that I wrote down at this point are thus: -5215, -199, -2079, 28963, 32738, -1126.

> Save to eeprom, power off, then on and launch the gui. At the values of bias equal to the ones written on paper?

Yes, after a new restart, the Calib Bias values are identical to the ones in the previous step. Final screenshot:

[image]

Thanks again and get well soon.

marcoaccame, 2019-01-11:
Very well.
Now next step: we attempt to operate on the offset of the amplifier.
Steps are:
1. Press ‘reset calib bias’. The adc should be again far away from zero. Save to eeprom now.
2. Press ‘tune amplifier’. The adc should be now close back to zero.
3. Write the new offset values. Save  to eeprom, power off, on again. Restart gui and check the offset values. Are they that same as you annotated?

gsaponaro, 2019-01-12:
1. After resetting the Calibration Bias and saving to EEPROM, we have:

[image]

2. After pressing Tune Amplifier, we have:

[image]

The offset of Channel 0 has changed a bit with respect to the previous step (from 792 to 745).

The new offset values at this point are: 745, 779, 793, 800, 803, 790. We save to EEPROM.

3. We do a power cycle of the motors and we restart the `FirmwareUpdater` GUI, getting:

[image]

Yes, the six offsets are the same that we annotated above.

marcoaccame, 2019-01-14:
Hi @gsaponaro, the behaviour of the board is not the one I expect. 
The button `Tune Amplifier` should automatically compute offsets which force the ADC values to be close to 0. In here, only channel 0 and maybe 1 are successful (it has ADC close to 0 and changes its offset) whereas the other channels have bad ADC values and the offset does not change at all.

We can do that now:
1. I am going to reproduce the set-up here in IIT to see if with a strain board of mine everything works fine.
2. pls arrange a skype + remote desktop session w/ @julijenv so that I can do some more tests with you (move the slider of some offsets, read the stdout on terminal, etc.).

gsaponaro, 2019-01-14:
Hi @marcoaccame , thank you for your insights and experiments. Indeed, it seems that with our STRAIN Channels 0 and 1 are successful, whereas Channels 2-5 "have bad ADC values and the offset does not change at all".

Some remarks:
* today I was going to try the latest tests again, but the following happened. Motors were freshly started. I ran `FirmwareUpdater --admin`, I selected the CFW2<1> network, it discovered the four boards (two BLL, one MC4, one STRAIN). I selected the STRAIN board (i.e., green arrow in the following screenshot), but...
[image]
... I saw these types of errors in the terminal (many of them):
```
QXcbConnection: XCB error: 2 (BadValue), sequence: 10595, resource id: 1368, major code: 130 (Unknown), minor code: 3
QXcbConnection: XCB error: 2 (BadValue), sequence: 10599, resource id: 1368, major code: 130 (Unknown), minor code: 3
QXcbConnection: XCB error: 2 (BadValue), sequence: 10603, resource id: 1368, major code: 130 (Unknown), minor code: 3
QXcbConnection: XCB error: 2 (BadValue), sequence: 10607, resource id: 1368, major code: 130 (Unknown), minor code: 3
```
Any idea about these errors?

* I restarted everything, this time I could select the board fine, and I could enter the calibration window. I re-did Reset Calib Bias, I saved to EEPROM, I pressed Tune Amplifier, and I observed the following offsets, very similar to before (except Channel 1 slightly changed): 745, **777**, 793, 800, 803, 790. I tried again a couple of times, the behaviour is consistent.

* I will try to schedule a session with @julijenv, as you suggest, next week (because this week I'm travelling). In the meantime, if you guys have any further suggestions, please let us know. Thanks.

marcoaccame, 2019-01-16:
Hi @gsaponaro, I have tested the behavior on our iCubGenova01 robot which has a CFW2 driver equal to your robot.

I could reproduce the problem and solve it.

The following commit solve the problem of incorrect loading of the calibration data file:  https://github.com/robotology/icub-main/commit/e3e8bf4772c1af39183db73f9a485a5fd11d82c4 

As you can read in the log, the CFW2 driver produces long bursts of CAN frames which the strain board loses when its RX queue gets full. It is the reason why the differences are in consecutive values (the last two calib bias and the first fullscale). A small delay between CAN frames has solved.

So, please: pull icub-main, recompile and try to load again the file  `calibrationDataSN094.txt`.
I am confident that now the values in EEPROM will be OK.

After that we shall look at the ADC acquisition values.

gsaponaro, 2019-01-20:
Hi @marcoaccame, great news: the fix worked, the calibration data file was correctly loaded. :+1:  
I saved it to EEPROM, and I verified (just for completeness) that the button `Save Calibration` indeed produces a file with the correct values.

Now the status is:

[image]

Looking forward to receiving your next instructions about ADC acquisition values. I will have access to the robot on Tuesday 22 afternoon. Thanks very much.

gsaponaro, 2019-01-22:
Hi @marcoaccame, I have tried to move the slider of channel 4 as follows.

Initial state:
[image]

Then I moved the slider to the left -> the values changed (slightly):
[image]

Then I moved the slider to the right -> the values changed significantly:
[image]

As instructed, I did not save to EEPROM.

marcoaccame, 2019-01-31:
Hi @gsaponaro,

from your test is seems to me that the strain board has a HW problem in channel 4 and quite surely also on channel 3. Maybe in the straingauges.

I think the channel 4 is broken because:
- the offset value from the calibration file should place the ADC value around 0. That happens with channels 0, 1, 2, 5 and also with the right strain (as from a previous picture).
- if you move the amplifier offset, the ADC value stays almost saturated in range [-32625, -27151].

So, i think you should replace the board. Please, contact @julijenv which can help you in either moving one from the leg to the left arm or to install a new one.

gsaponaro, 2019-01-31:
@marcoaccame, thank you for the analysis. It seems that this strain board is KO, then.

@julijenv, we would like use a strain board from one of the legs, and put it in the arm, for proceeding with our experiments. Could you explain us the necessary steps? (We also want to purchase a new board, but that's going to take longer.)

julijenv, 2019-02-07:
Hi @gsaponaro ,
Please follow the step by step following guide to remove and replace the FT sensor from the leg to put on the upper arm. (advice : please take some pictures to make sure the FT sensor side and connector get out)
1. Remove the covers of the thigh
2. remove these 4 screws (V4x10)
[image]
3. remove those 8 screws (V4x10)
[image]
At this point you will have just to remove the screws holding the FT from the top and bottom mechanical part (resp V3x12 and V3x8)
[image]
[image]
The easiest part is now done.
Let's see how to remove the upper arm FT sensor now:
1. Remove the covers of the upper arm.
2. Removing the forearm: this step require first that you disconnect the overall connector coming from the bread of wires at the level of the soulder-upper arm that link the 3 MC4 boards beneath the upper arm.
At this point you should have the pack of 3 boards hanging by the bread going to the forearm.
[image]
3. remove the 3 screws holding the foream attach to the elbow. (V3x10)
[image]

Here is the overall mechanical pieces for the upper arm
[image]
1. try to detect this mechanical assembly
[image]
2. Remove the 6 screws around the circle top part.(V3x12)
[image]
3. Remove the 6 screws (V3x8)
[image]
4. Remove the 6 screws (V3x8)
[image]
4. Remove the 6 screws (V3x14)
[image]
And here the result
[image]

swap the FT and redo every step in the other way around.
Let me know,
best of luck,

julijenv, 2019-03-06:
Hi @gsaponaro , did you succeed to replace the FT sensor??

gsaponaro, 2019-03-06:
Hi @julijenv, we have not done it yet due to some internal delays and travels... Thank you for your instructions, they were very clear, I'll let you know how it goes asap.

gsaponaro, 2019-03-25:
Hi, we took the F/T sensor that was in the left left (SN147)...

[image]

... and we inserted it in the left arm (replacing the faulty sensor), following @julijenv 's instructions. Thanks a lot.

Now the CAN1 network of the left arm in `FirmwareUpdater` is as follows:

[image]

[image]

```bash
$ yarp read ... /icub/left_arm/analog:o
yarp: Port /tmp/port/1 active at tcp://10.10.1.53:10051/
yarp: Receiving input from /icub/left_arm/analog:o to /tmp/port/1 using tcp
78.2936096191406 122.499267578125 -45.2840881347656 1.094970703125 -2.00845336914062 0.52978515625
```

Questions for @marcoaccame : do we need to re-calibrate this sensor, which was moved from leg to arm? If so, can you tell us the steps?

I have not run yet `demoForceControl`, for safety considerations.

julijenv, 2019-04-03:
@marcoaccame , any feedback to give???

gsaponaro, 2019-04-09:
Excellent, thanks.

closed 2019-04-09 by gsaponaro