robotology/mesh-support #753: yarprobotinterface not starting [iCubManchester01]

samvinanzi, 2018-12-20:
### Description of the failure

Last night we have been working with our iCub and everything was fine. We didn't perform any updates, just regular use. This morning we couldn't start yarprobotinterface because of a CAN interface problem.

### Detailed conditions and logs of the failure

Here is the log of the failure:

```
icub@pc104:~$ yarprobotinterface
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./cartesian/left_arm_cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./cartesian/right_arm_cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/left_arm_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/right_arm_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/left_leg_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/right_leg_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/head_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/motorControl/torso_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_left_arm.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_left_hand.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_right_arm.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_right_hand.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_left_leg.xml 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_left_leg.xml at line 5 . Trying to import a file for the wrong robot. Found iCubLisboa01 instead of iCubManchester01 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_right_leg.xml 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_right_leg.xml at line 5 . Trying to import a file for the wrong robot. Found iCubLisboa01 instead of iCubManchester01 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_head.xml 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_head.xml at line 2 . "devices" element should contain the "robot" attribute 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_head.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of iCubManchester01 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_head.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_head.xml at line 2 . Import a file for a different robot build. Found 0 instead of 1 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/motorControl/icub_torso.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/VFT/left_arm_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/VFT/left_leg_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/VFT/right_arm_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/VFT/right_leg_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/VFT/torso_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/VFT/left_arm_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/VFT/left_leg_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/VFT/right_arm_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/VFT/right_leg_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/VFT/torso_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/FT/left_arm_FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/FT/left_leg_FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/FT/right_arm_FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/FT/right_leg_FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/MAIS/left_hand_mais_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./wrappers/MAIS/right_hand_mais_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/FT/left_arm_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/FT/left_leg_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/FT/right_arm_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubManchester01/./hardware/FT/right_leg_strain.xml 
[... 677 lines cut ...]
[INFO]created device <xsensmtx>. See C++ class yarp::dev::XSensMTx for documentation.
[... cut ...]
[INFO]created device <parametricCalibrator>. See C++ class yarp::dev::parametricCalibrator for documentation.
[INFO]created device <parametricCalibrator>. See C++ class yarp::dev::parametricCalibrator for documentation.
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
Segmentation fault
```

We would greatly appreciate some support in tracking down the problem.
Thank you!

julijenv, 2018-12-20:
Hi @samvinanzi ,
from this following warning messages from your log file:```
[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.6
[INFO]Firmware report:
[INFO]sharedcan [3] joint: 0 can_address:  5 board type: 4 (BLL) version: 1.51 build: 87 CAN_protocol:1.6
[INFO]sharedcan [3] joint: 1 can_address:  5 board type: 4 (BLL) version: 1.51 build: 87 CAN_protocol:1.6
[INFO]sharedcan [3] joint: 2 can_address:  6 board type: 4 (BLL) version: 1.51 build: 87 CAN_protocol:1.6
[INFO]sharedcan [3] joint: 3 can_address:  6 board type: 4 (BLL) version: 1.51 build: 87 CAN_protocol:1.6
[WARNING]sharedcan [3] joint: 4 can_address:  7 Unable to detect firmware version. Old firmware running?
[WARNING]sharedcan [3] joint: 5 can_address:  7 Unable to detect firmware version. Old firmware running?
[INFO]**********************************
[INFO]
```
I can conclude that the last left leg control board (BLL) is not responding correctly.
Could you check with the Firmwareupdater if you can see 3 BLL boards on the net3 after having click on discover.
To have something comparable you can also check the net4 (right leg) and it should be exactly the same as the left leg boards that you should detect apart for the ID number., could you please take a screenshot for net 3 and 4?
if you cant see 3 boards you may have a wire broken or disconnected net3 at the level of the left leg,
let me know

samvinanzi, 2018-12-20:
Thanks for the quick response.
Here is the screenshot:

[image]

We also have investigated the wiring of the leg, just to be sure, but it seems ok (we have pictures if needed).

julijenv, 2018-12-20:
Hi @samvinanzi ,
ok easy the pblm is that the BLL ID has changed for some reason - it happened not frequently, but it is something already seen in the past-
You just need to change the ID number from 15 to 7 - 7 is the good ID.
Let me know

samvinanzi, 2018-12-20:
May I ask how to perform the modification? There is a "Change Info" button but it's not very clear.

julijenv, 2018-12-20:
could you be tomorrow on skype : my ID is icub.support and try to call me (install teamviewer please)

samvinanzi, 2018-12-20:
Thank you, we solved it and `yarprobotinterface` is now running.

**For any further reference**, we modified the board ID by running `canLoader`, double-clicking on the specific ID and typing in the correct value.

Many thanks for the fast and efficient help.

closed 2018-12-20 by samvinanzi