robotology/mesh-support #755: Robot goes in hardware fault when using VOCAB_IM_COMPLIANT motorControlImpedance

caicaixia, 2019-01-08:
Dear Sir,

Today I tried the tutorial demo motorControlImpedance for the impedance control of iCub.

Its is very simple demo:
```
if (times%2)
        {
             // set the elbow joint only in compliant mode
             ictrl->setPositionMode(3);
             iint->setInteractionMode(3,VOCAB_IM_COMPLIANT);
             // set new reference positions
             command[0]=-50;
             command[1]=20;
             command[2]=-10;
             command[3]=60;
        }
        else
        {
             // set the elbow joint in stiff mode
            ictrl->setPositionMode(3);
             iint->setInteractionMode(3,VOCAB_IM_STIFF);
             // set new reference positions
             command[0]=-20;
             command[1]=40;
             command[2]=-10;
             command[3]=30;  
        }
```

However I have few problems:
1: After icub went to the `VOCAB_IM_COMPLIANT`, then the elbow joint will show a hardware fault. Then even it changes back to `VOCAB_IM_STIFF` mode, its seems like joint3 is not able to move again.

[image]

2: What is the mean of new reference positions? Joint3 will go to the new reference position or not? From the demo, I saw the command[3]=60; or command[3]=30; are never reached, it seems like the elbow joint is not controlled. This is maybe due to the problem 1;

julijenv, 2019-01-08:
hi @caicaixia ,
could please tell me which robot your are using?

caicaixia, 2019-01-08:
Hello @julijenv 

1:  I am working in iCubSingapore01;
2:  I tried to move the elbow joint from the 2 extreme values with the sliding bar, its no problem;
3:  I also tested other joints for  the `VOCAB_IM_COMPLIANT` mode:
     `iint->setInteractionMode(1,VOCAB_IM_COMPLIANT);`

I set joint 1,  then joint 0,1,2  all become compliant, i guess this is because those 3 joints are belong to shoulder togehter.   This test has the same hardware fault problem after the `VOCAB_IM_COMPLIANT` mode.
[image]

julijenv, 2019-01-08:
Hello @caicaixia ,
I'm not the expert of the software part but I believe that it is the way you use the function 

> motorControlImpedance

 that lead to this problem, I will let my colleague @pattacini to get an eye to that. But I fear you may need to insert the entire code of the while loop inside a reply to get more inside of the problem you may face and giving you the right tip to use the function.
Please consider copy paste your entire code for the use of the function to help the resolution.
You may alternatively send the file you use.
thx in advance