robotology/mesh-support #756: Problem with icub left hand middle finger

dhgarcia, 2019-01-08:
### Description of the failure

The iCub left hand middle finger doesn't move with yarpmotorgui joint 13 (l_middle_proximal).
Additionally, the fingersTuner doesn't run.
We don't know if it is related.

Everything else runs fine for iCubStartup (yarprobotinterface, iKinCartesianSolver, iKinGazeCtrl, wholeBodyDynamics, gravityCompensator, imuFilter)

### Detailed conditions and logs of the failure

Every other joint for both hands move normally through the yarpmotorgui except joint 13 (l_middle_proximal)

We try to run the fingersTuner and it fails with the following message:
[ERROR]group [left_hand] is missing!
[INFO]RFModule failed to open.

How can we debug the issue?

julijenv, 2019-01-09:
Hi @dhgarcia ,
first of all, happy new year,
second, did you make any change lately on your config files or did you update icub and the yarp repositories?!?!?! if not it is more probably a hardware failure... Let's investigate this part:
I am not particulary familiar with the fingersTuner module or function, so I cant tell you what it is supposed to tell you when everything (hardwarly speaking) is working fine BUT I will try to identify with you the state of hardware sanity.
First of everything please read carefully my explanations before starting debugging. LEt's start.
Please make sure iCub is not at work (shutdown the motor switch at least - safety first), take a flat screwdriver not too thin not too big , it has to fit into the screw of the motor pulley that you can find at the level of the elbow (beginning of the forearm) - You should be able to see 3 square shape insde of which you could identify 3 flat screw heads so as the picture below.
[image]
The middle one is the prox thumb motor (joint 9) - the left and right ones are the middle and index proximal motor (j11 and j13).
Please check by removing the cover of the forearm if the wire inside the pulley looks broken, or alternatively, can you move the middle finger prox joint just by moving it up and down eith your hand.
next, ), by rotating the screw you will be able to move up and down the fingers ( watch out - the motor are not always easily backdrivable, please rotate it with care if it is HARD, DO NOT FORCE, test another joint instead and find out what should be the minimum force to move the joint)
let me know your findings, you can semd me pictures of what looks odd to you.

dhgarcia, 2019-01-09:
Hi @julijenv Happy New Year!! 
Thank you for your reply and the instructions.

We inspected the arm after removing the cover and it all look alright.
At the beginning we could not move the finger by pushing it by hand, but after moving it by rotating the motor with screw driver (it was not stiff in that case) several times it started to move also when finger pushed by hand.
After doing  that we found that now we can move the finger using yarpmotorgui (joint 13 moves correctly now).

The issue with the fingersTuner it is still the same.
We can close this issue and open a new one for the fingersTuner if you think is best.

julijenv, 2019-01-09:
Hi @dhgarcia ,
if now you can move every single joint of the hands (left and right) I think it would be better to close this issue and open a specific one on QA for this fingersTuner. Which is of course more software related.

closed 2019-01-10 by dhgarcia