robotology/mesh-support #760: Robot errors when run YARPROBOTINTERFACE

caicaixia, 2019-01-16:
Hello guys,

Today we are doing experiments with iCub.  At beginning everything is ok.  Then we restart iCub after sometime, it can not run anymore.

When run `yarprobotinterface`, it shows many errors, and the robot can not connected anymore.
[image]

[image]

what can we do now?

caicaixia, 2019-01-16:
@julijenv , thanks very much. We will try to solve this problem after our demo next week.  Thanks again for your time.

julijenv, 2019-02-27:
Hi @tasbolat1  and @caicaixia ,
Did you succeed to repair this cable?
let us know

closed 2019-10-02 by maggia80