robotology/mesh-support #761: Left shoulder yaw cable broke during motor calibration tests

nunoguedelha, 2019-01-16:
### Description of the failure
While performing the identification of motor PWM to current model parameters on the left shoulder motors, the cable actuating the shoulder yaw broke.

### Detailed conditions and logs of the failure
More precisely, the procedure consisted in applying an external torque to the targeted joint while the joint was controlled in position (fixed).

nunoguedelha, 2019-01-16:
CC @julijenv

nunoguedelha, 2019-01-17:
After @julijenv fixed the shoulder cable, I have performed some tests moving the 3 joints with the `yarpmotorgui`. Everything is working fine. I let @julijenv close the issue. Thanks!

julijenv, 2019-01-18:
It has been used the following items to restore the orobt
ITTCODE Qta Dexscription
4765 x 1 : shoulder cable 8
4758 x 1 : shoulder cable 1

closed 2019-01-18 by julijenv