robotology/mesh-support #762: [R1-mk2] Issues with the hand skin sensors

jchevrie, 2019-01-21:
### Description of the failure

The skin sensors on the R1-mk2 hands are not working properly:

- some possible hardware issues on the distal fingers (largest part): the sensors readings are noisy and seem to sometimes cause a reset of the board.
Due to the randomness of the values it may be a problem with a wire connection.

- there seem to be some mapping errors in the `ICubSkinGui`, since the pressed sensors are not always displayed at the correct position in the gui.

### Detailed conditions and logs of the failure

At first the sensors readings seemed weird for almost all sensors when looking at the gui.
Changing some parameters (`skinType` and `shift`) that were incorrect in the config files (`right_hand-mc7-skinSpec.xml` and `left_hand-mc9-skinSpec.xml`) solved the issue for some sensors.
But the noisy distal fingers sensors are still not behaving correctly.

[image]

ale-git, 2019-01-21:
Hi @jchevrie, are the sensors swapped inside the same patch or exchanged between different patches?

jchevrie, 2019-01-21:
Hi @ale-git 
They are swapped inside the same patch only.
Only a few seem actually swapped, but the reason why we think there are mapping issues is that some sensors never light up in the gui.

closed 2019-11-24 by stale[bot]