robotology/mesh-support #763: [Genova02] imu not working

arrenglover, 2019-01-23:
### Description of the failure

yarprobotinterface fails when trying to connect to the head imu. it is reproducible and the robot only works when the imu is commented from the icub_all.ini

### Detailed conditions and logs of the failure

[ERROR]Bosch BNO055 IMU - Received unknown response message. 
            If this error happens more than once in a row, please check serial communication is working fine and it isn't affected by electrical disturbance.

julijenv, 2019-01-24:
Please see [this](https://redmine.robotology.eu/easy_knowledge_stories/56) .
@Nicogene , by typing uname -r : I got v4.15 but unfortunately your tar.gz doesnt contain this version is there any work around?!?

Nicogene, 2019-01-25:
I'd try with the closest one(`4.11`), if it doesn't work we have to ask to @maggia80 to ask to the manufacturer the kernel modules for newer versions of the kernel.

julijenv, 2019-01-25:
We try yesterdady but there was an error message when it compiled.
normally @Iaxama should provide us with the plain printscreen of the error message.

Nicogene, 2019-01-25:
I think that the only way it is to contact the vendor and ask if they have the kernel module for the kernel 4.15 or greater, cc @maggia80 .

Nicogene, 2019-01-25:
https://github.com/robotology/robots-configuration/pull/83/commits/c52e7b4c0b75a06bdc428f949bd1472017fd0c66 
You need to change this to run the IMU with  i2c bus configuration.

Nicogene, 2019-01-25:
The kernel modules that you are trying to build and install are needed **if and only if** the IMU is attached to the i2c bus(new `icub-head`, like the one of the `iCubGenova04`).
But also in this case they should be already installed in the image provided by Matteo if needed.

I'm quite sure that the `iCubGenova02`'s `icub-head` is an old one, the IMU should be attached to the serial bus, then forget

> [robotology/robots-configuration@c52e7b4](https://github.com/robotology/robots-configuration/commit/c52e7b4c0b75a06bdc428f949bd1472017fd0c66)
> You need to change this to run the IMU with i2c bus configuration.

and

> I think that the only way it is to contact the vendor and ask if they have the kernel module for the kernel 4.15 or greater, cc @maggia80 .

Sorry I misunderstood the problem, I thought you had issues with the i2c configuration but I think it is a problem of the serial bus.

@barbalberto wrote the serial part, maybe he has more insight.

julijenv, 2019-01-25:
Hi @Nicogene , 
the head of iCubGenova02 - Purple - has been updated and by so the bosch sensor is attached to the i2C port.

julijenv, 2019-01-25:
Hi @mbrunettini ,
it seems that the Kernell v4.15 of the head, prevent us to install the i2C module to drive the bosch sensor.
Could you give us an hand on that please,
thx in advance

arrenglover, 2019-01-25:
I tried this command:
`sudo bin/yarpdev --device imuBosch_BNO055 --i2c /dev/i2c-2`
and managed to open the imu device correctly.

note that it was found on the i2c-2 and not 1.

an additional problem is that i need to be sudo to access the i2c devices. I can solve this by adding the `icub` user to the i2c group, but this is something that probably should be added to the install image - @mbrunettini

Nicogene, 2019-01-25:
yes the command is 
```bash
$ usermod -a -G i2c icub
```
So now it is working?
Now I understand the error of the device, it printed that message because nothing was attached to the serial bus.

Nicogene, 2019-01-25:
> note that it was found on the i2c-2 and not 1.

`i2c-1` was for the `iCubGenova04`, it could be different 
the way to discover is : 
```
$ i2c-detect -l
```

arrenglover, 2019-01-25:
I spoke to Matteo in person and he said there was something that should be done to make sure the i2c device is always configured for the same i2c line (i.e. i2c-1). He should be able to configure the i2c correctly next week and we can do a final test to make sure the yarprobotinterface correctly opens the imu device.

traversaro, 2019-01-25:
> > note that it was found on the i2c-2 and not 1.
>
> `i2c-1` was for the `iCubGenova04`, it could be different 
the way to discover is : 
> ```
> $ i2c-detect -l
> ```

According to Linux kernel documentation, the specific filename for a i2c device can change as soon as you reboot the machine, see https://github.com/torvalds/linux/blob/6f0d349d922ba44e4348a17a78ea51b7135965b1/Documentation/i2c/dev-interface#L28 : 
> Now, you have to decide which adapter you want to access. You should
> inspect /sys/class/i2c-dev/ or run "i2cdetect -l" to decide this.
> Adapter numbers are assigned somewhat dynamically, so you can not
> assume much about them. They can even change from one boot to the next.

That's the reason you need to add a[ `udev` rule](https://www.freedesktop.org/software/systemd/man/udev.html) to ensure that there is a consistent symlink with a permanent name, and that is what @arrenglover was mentioning.

mbrunettini, 2019-01-28:
to create the UDEV rule, the output of the output of the following command will be really helpful:
`udevadm info -a -n /dev/i2c-2`
(or` /dev/i2c-1`)
Thanks!

arrenglover, 2019-01-28:
I added icub user to the group so I can connect to the i2c device without sudo, but I don't know exactly what I should be doing with udev. Is there an example I can copy in?
P.s. I'm unable to test the yarprobotinterface starts the IMU correctly as the motors are not working at the moment.

arrenglover, 2019-01-30:
1. the IMU is connecting when running the yarprobotinterface but the yarpdev constantly outputs a warning message:
--
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
[WARNING]dcm2rpy() in singularity: choosing one solution among multiple
--
This warning message goes away on the second opening of the device, but seems like there is something not configured correctly.

2. @mbrunettini Can I please have some help of how to set the udev rule

mbrunettini, 2019-01-30:
> to create the UDEV rule, the output of the output of the following command will be really helpful:
> udevadm info -a -n /dev/i2c-2
> (or /dev/i2c-1)
> Thanks!

arrenglover, 2019-01-30:
The imu is currently on /dev/i2c-2 but it can be put on /dev/i2c-1 to make it consistent with icubgenova04. Output of `udevadm info -a -n /dev/i2c-2`

> looking at device '/devices/platform/kempld/kempld-i2c/i2c-2/i2c-dev/i2c-2':
>     KERNEL=="i2c-2"
>     SUBSYSTEM=="i2c-dev"
>     DRIVER==""
>     ATTR{name}=="i2c-kempld"
> 
>   looking at parent device '/devices/platform/kempld/kempld-i2c/i2c-2':
>     KERNELS=="i2c-2"
>     SUBSYSTEMS=="i2c"
>     DRIVERS==""
>     ATTRS{name}=="i2c-kempld"
> 
>   looking at parent device '/devices/platform/kempld/kempld-i2c':
>     KERNELS=="kempld-i2c"
>     SUBSYSTEMS=="platform"
>     DRIVERS=="kempld-i2c"
>     ATTRS{driver_override}=="(null)"
> 
>   looking at parent device '/devices/platform/kempld':
>     KERNELS=="kempld"
>     SUBSYSTEMS=="platform"
>     DRIVERS=="kempld"
>     ATTRS{driver_override}=="(null)"
>     ATTRS{pld_specification}=="2.5"
>     ATTRS{pld_type}=="release"
>     ATTRS{pld_version}=="P206.002B"
> 
>   looking at parent device '/devices/platform':
>     KERNELS=="platform"
>     SUBSYSTEMS==""
>     DRIVERS==""

Output of `udevadm info -a -n /dev/i2c-1`

>   looking at device '/devices/pci0000:00/0000:00:1f.4/i2c-1/i2c-dev/i2c-1':
>     KERNEL=="i2c-1"
>     SUBSYSTEM=="i2c-dev"
>     DRIVER==""
>     ATTR{name}=="SMBus I801 adapter at f040"
> 
>   looking at parent device '/devices/pci0000:00/0000:00:1f.4/i2c-1':
>     KERNELS=="i2c-1"
>     SUBSYSTEMS=="i2c"
>     DRIVERS==""
>     ATTRS{name}=="SMBus I801 adapter at f040"
> 
>   looking at parent device '/devices/pci0000:00/0000:00:1f.4':
>     KERNELS=="0000:00:1f.4"
>     SUBSYSTEMS=="pci"
>     DRIVERS=="i801_smbus"
>     ATTRS{broken_parity_status}=="0"
>     ATTRS{class}=="0x0c0500"
>     ATTRS{consistent_dma_mask_bits}=="32"
>     ATTRS{d3cold_allowed}=="0"
>     ATTRS{device}=="0x9d23"
>     ATTRS{dma_mask_bits}=="32"
>     ATTRS{driver_override}=="(null)"
>     ATTRS{enable}=="1"
>     ATTRS{irq}=="16"
>     ATTRS{local_cpulist}=="0-3"
>     ATTRS{local_cpus}=="f"
>     ATTRS{msi_bus}=="1"
>     ATTRS{numa_node}=="-1"
>     ATTRS{revision}=="0x21"
>     ATTRS{subsystem_device}=="0x7270"
>     ATTRS{subsystem_vendor}=="0x8086"
>     ATTRS{vendor}=="0x8086"
> 
>   looking at parent device '/devices/pci0000:00':
>     KERNELS=="pci0000:00"
>     SUBSYSTEMS==""
>     DRIVERS==""

mbrunettini, 2019-02-01:
Based on the info @arrenglover  provided (thanks!), I added the file `/etc/udev/rules.d\88-bosch-i2c-imu.rules` as follows:
` /etc/udev/rules.d/88-bosch-i2c-imu.rules

# iCub Bosch i2c IMU
KERNEL=="i2c-2", SUBSYSTEM=="i2c-dev", ATTRS{name}=="i2c-kempld", SYMLINK+="bosch-i2c-imu", MODE="0660", GROUP="i2c"`

This creates a symlink to the correct /etc/i2c device called

 `/dev/bosch-i2c-imu`

You now need to reconfigure the application to point to the above link.
This fix is already in place on the Purple and Green, and Green users confirm that is working

arrenglover, 2019-02-01:
Thanks Matteo. I see the device is `/dev/icub-i2c-imu' which is slightly different to what you wrote. To keep things consistent, which one is correct?

mbrunettini, 2019-02-01:
/dev/bosch-i2c-imu

arrenglover, 2019-02-01:
I updated the udev file to SYMLINK to `/dev/bosch-i2c-imu' and updated the configuration script`hardware/inertials/head_intertial.xml' to point to `dev/bosch-i2c-imu'.

The imu is now successfully opened by the yarprobotinterface and the data can be read off the port.

arrenglover, 2019-02-04:
@mbrunettini today when turning on the robot the imu device had moved from "i2c-2" to "i2c-1" and therefore the udev rule could not complete and I didn't get the symlink. Should the KERNEL="i2c-2" line be removed from the udev?

mbrunettini, 2019-02-04:
please try to replace the part
`"KERNEL="i2c-2`
with
`ATTR{name}="i2c-kempld"`
and let me know

mbrunettini, 2019-02-04:
moreover, can you post here the output of 
`udevadm info -a -p  $(udevadm info -q path -n  /dev/i2c-1)`

arrenglover, 2019-02-05:
I removed `KERNEL="i2c-2"` and the symlink was still created correctly. If you are okay with this fix, please close the issue again.

mbrunettini, 2019-02-05:
Good, closing

closed 2019-02-05 by mbrunettini