robotology/mesh-support #764: [genova02] motors not working

arrenglover, 2019-01-25:
### Description of the failure

when trying to start the robot the motor light is blinking red and green.

### Detailed conditions and logs of the failure

We were using the robot for some time, and when trying to perform a restart the motors no longer will turn on.

julijenv, 2019-01-25:
Hi @arrenglover ,
I've just been there and I tried to launched the head and motor, everything went smooth.
could you please try to make a video of the blinking when this happen again, thx in advance.
Maybe the board control the power line went out of its mind, but we need to understand why.

arrenglover, 2019-01-25:
thanks @julijenv. Yep, I was able to turn them on again now as well. Could it be a heating component? We lost connection to the icub-zynq, which was the reason to do the reset. Perhaps the zynq overheated and caused problems for the power supply in the back-pack?

I'll try to gather more information next time it happens.

arrenglover, 2019-01-28:
@julijenv the motors are in fault from initially turning it on after the weekend.

julijenv, 2019-01-29:
Hi @arrenglover ,
could you please upload a video on LED fault, it will help to understand ? thx in advance

julijenv, 2019-02-06:
Hi @arrenglover ,
@MrAndrea readjust the voltage of the power supply that was set by 20 % higher than normal usage due to specidfic request from the jumping team.
Now so far I dont think we have been seeing the error again.
keep on hold, I will close later.

arrenglover, 2019-02-19:
@julijenv - we haven't seen this issue for some weeks, I'm happy to close for now if you are.

closed 2019-02-20 by julijenv