robotology/mesh-support #765: [Genova02] Eye version/vergence calibration

arrenglover, 2019-01-25:
### Description of the failure

Calibration of the eye version and vergence in the calibrators/icub-head.xml changes the left eye and right eye individually instead.

### Detailed conditions and logs of the failure

This is not so much a failure, but it is not documented anywhere, so I thought to put a note here. However, calibrating the left eye and right eye individually is currently more difficult as the yarpmotorgui does not allow to control the eyes individually. It resulted in having to restart the yarprobotinterface multiple times to check and change the value in the file.

arrenglover, 2019-01-30:
@valegagge - do you have any further information about this change to the calibration?

julijenv, 2019-02-06:
@valegagge , could you try to help him in this matter. thx in advance

closed 2019-10-12 by pattacini