robotology/mesh-support #766: [genova02] zynq crashing (overheating?)

arrenglover, 2019-01-25:
### Description of the failure

After running the robot for an extended time (some hours) we lose connection to the zynq, requiring a restart of the robot.

### Detailed conditions and logs of the failure

It appears to be an overheating issue as we have had it before, the fans of the new head are now further away from the zynq-board and we do not have a specific fan used to cool the zynq as we had before.

seems related to https://github.com/robotology/icub-support/issues/764 as it is when we restart the robot that the back-pack is in error and does not work immediately.

arrenglover, 2019-02-01:
the problem could also be related to a large amount of events causing the driver to continually output a kernel warning message and the zynq locks up. This is fixed in the latest version of the fpga and driver code.

maggia80, 2019-10-01:
@arrenglover can we close this issue?
 
ciao,
Marco

arrenglover, 2019-10-01:
@maggia80 yes. I am going to open a new issue for a different problem instead :stuck_out_tongue_closed_eyes:

maggia80, 2019-10-02:
ehehe