robotology/mesh-support #768: Problem with xml version 3.0 in yarprobotinterface

frank-foerster, 2019-01-28:
After many months I updated both yarp and all icub-related software for and on our robot, including the installation of the most recent live image. 
When trying to start the `yarprobotinterface` I get now the following error message: 

```
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml at line 2 . Unknown document type. Missing version in Url http://www.icub.org/DTD/yarprobotinterfaceV3.0.dtd
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV1.0
[DEBUG]yarprobotinterface: using xml parser for DTD v1.x 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml 
[FATAL]Syntax error while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml at line 6 . Expected "device" or "devices". Found xi:include
```
etc. pp.

So apparently the interface doesn't like the 2nd line, \<!DOCTYPE ...\> ... , with the new V3.0 DTD URL.
After checking the issues on Github, and finding nothing, I noticed that most `icub_all.xml' files had been changed in July to conform with some new XML format. Having looked at the same files of some of the other robots on here, I noticed that some of you removed this very line from their respective icub_all.xml files. So I tried that, but then I get a 

```
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV1.0 
[DEBUG]yarprobotinterface: using xml parser for DTD v1.x 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml 
[FATAL]Syntax error while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./icub_all.xml at line 6 . Expected "device" or "devices". Found xi:include
```

Do I need to copy this non-existent *.dtd file to somewhere, or pass the interface some parameter? 
What am I missing?

randaz81, 2019-01-28:
There was a typo in the xml, sorry for the issue.
fixed by https://github.com/robotology/robots-configuration/commit/de8f17bef917272f26fe83d40c68c411b17105b9

closed 2019-01-28 by randaz81

frank-foerster, 2019-01-28:
Thank you very much! Just tried it and it works now.