robotology/mesh-support #769: request for network file for firmware upgrade

frank-foerster, 2019-01-28:
### Description ~~of the failure~~
Not really a failure, but a request for the provision of the file `_**network.iCubHertfordshire01**_' in \$ROBOT_CODE/robot-configurations/iCubHertfordshire01 as this file is needed for a firmware upgrade.

### Detailed conditions ~~and logs of the failure~~
We just tried to conduct a firmware upgrade with the (relatively new) _FirmwareUpdater_ and discovered that I need the abovementioned file in order to do this. The doc file prompts me to request that file, so here we go ... .

Many thanks in advance

marcoaccame, 2019-01-29:
Hi @frank-foerster,
thanks for having read the doc file and having posted this request for the `network.iCubHertfordshire01.xml` file.

The file contains a description of all the CFW2 networks of your robot so that you can launch an automated script for the firmware exploration and update operations, but as you probably already know, you can still update the firmware of you boards with the FirmwareUpdater GUI in interactive mode.

Having clarified that, ... **yes: I will produce the file `network.iCubHertfordshire01.xml` for your robot very shortly. I hope by today or tomorrow**. 

I will produce the file on the basis of what is written in your xml files under the `robots-configuration/iCubHertfordshire01/hardware` folder.

However, I would like to be 100% sure that the xml files are a correct picture of your robot, hence ...  please would you help me in the following way?
1. Power on the motors.
2. Launch the FirmwareUpdater GUI and make sure that you can see all the drivers listed in your `robots-configuration/iCubHertfordshire01/firmwareupdater.ini` file.  They should be CFW2 drivers from 0 to 9. 
3. One at a time, select each CFW2 driver and then click on button `Discover` to explore what's inside. 
4. Enlarge the board area to be sure we can read all the information of every CAN board, do a printscreen and post the picture in this issue.

In this way, you can test the correctness of the FirmwareUpdater tool which will be used in non interactive mode later on by the bash script, and also I can double check the content of the `network.iCubHertfordshire01.xml` file.

cc @julijenv: pls can you mark this issue as a reference for future production of `network.$YARP_ROBOT_NAME.xml` files? thanx.
cc @davidetome: I will need to commit the file to master...

frank-foerster, 2019-01-29:
HI @marcoaccame ,

as it happens I just took a screenshot yesterday in order to be able to get a before- and after picture once the update is performed. 
Here it is: 
[image]

Many thanks for the help!

marcoaccame, 2019-01-29:
very good. give me 30 minutes and i will send you the file to test.

marcoaccame, 2019-01-29:
Here is a **tentative** `network.iCubHertfordshire01.xml` file: 
[network.iCubHertfordshire01.xml.txt](https://github.com/robotology/icub-support/files/2807325/network.iCubHertfordshire01.xml.txt)

I would like you to test it before we are sure it is correct.

So please:
1. Copy it in folder `robots-configuration/iCubHertfordshire01` (remove the .txt extension first ...)
2. Go to folder `icub-firmware-build/scripts`
3. Run the run script `./FirmwareUpdater.script.info.all.sh`. It will check the correct syntax of the network xml file i sent you by printing all info in its inside.
4. Do copies of the file `FirmwareUpdater.script.verify.bodypart.sh.txt` (without the .txt) renaming it with the relevant bodypart name you want to test (head, left_leg, right_leg, right_arm, left_arm, torso) and also edit each of them so that it contains the correct  name instead of the bodypart string.
3. Run each of the scripts one after the other. For example, run: `./FirmwareUpdater.script.verify.head.sh`. It will verify if all the boards described by the network xml file and belonging to the head are present and if their firmware versions are coherent with what reported by `icub-firmware-build/info/firmware.info.xml`.

[FirmwareUpdater.script.verify.bodypart.sh.txt](https://github.com/robotology/icub-support/files/2807356/FirmwareUpdater.script.verify.bodypart.sh.txt)

pls send me the log files back to me.

frank-foerster, 2019-01-29:
Hi,

so he output of `./FirmwareUpdater.script.info.all.sh` looks good to me, but all off the log-files of he 
`./FirmwareUpdater.script.verify.*.sh` scripts contain errors to the effect that the required boards could not be found. 
Note, that I have switched on the motors / motor power supply of the iCub, but I didn't do anything else (trying to startup yarprobotinterface or the like).

Here are the logs:

FirmwareUpdater.script.info.all.sh.log:
[FirmwareUpdater.script.info.all.sh.log](https://github.com/robotology/icub-support/files/2808712/FirmwareUpdater.script.info.all.sh.log)

FirmwareUpdater*verify* logs:
[log.of.FirmwareUpdater.iCubHertfordshire01.verify.head.txt](https://github.com/robotology/icub-support/files/2808713/log.of.FirmwareUpdater.iCubHertfordshire01.verify.head.txt)

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.left_arm.txt](https://github.com/robotology/icub-support/files/2808714/log.of.FirmwareUpdater.iCubHertfordshire01.verify.left_arm.txt)

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.right_arm.txt](https://github.com/robotology/icub-support/files/2808717/log.of.FirmwareUpdater.iCubHertfordshire01.verify.right_arm.txt)

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.left_leg.txt](https://github.com/robotology/icub-support/files/2808718/log.of.FirmwareUpdater.iCubHertfordshire01.verify.left_leg.txt)

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.right_leg.txt](https://github.com/robotology/icub-support/files/2808719/log.of.FirmwareUpdater.iCubHertfordshire01.verify.right_leg.txt)

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.torso.txt](https://github.com/robotology/icub-support/files/2808720/log.of.FirmwareUpdater.iCubHertfordshire01.verify.torso.txt)

Thanks!

marcoaccame, 2019-01-30:
Hi @frank-foerster,

the error in FirmwareUpdater.script.info.all.sh.log is due to the fact that the file `info\firmware.info.xml` did not contain an entry for a bll with a given firmware version. I have added it. I attach the new file: please test it.

[firmware.info.xml.txt](https://github.com/robotology/icub-support/files/2811721/firmware.info.xml.txt)

The other errors show that the FirmwareUpdater cannot find any board on the CFW driver. But you have already fond boards beneath with the GUI, so it may be that the driver was busy. 

We do now a further test. Please try that:

1. switch motors on and then run `FirmwareUpdater` in GUI mode and explore the CFW2<0> bus. make sure you see the four boards that are shown in your previous screenshot.
2. **close** the `FirmwareUpdater` GUI.
3. edit the script of the head it so that we add some extra verbosity: change `-p head -a verify  | tee` into `-p head -a verify -v 3 | tee` 
4. run the script of the head again.

In the meantime, I will try once more the script on our robot iCubGenova01 which has exactly the same boards on the head.

frank-foerster, 2019-01-30:
Hi @marcoaccame 

So, I replaced our local `firmware.info.xml` with the one you sent me, re-ran `FirmwareUpdater.script.info.all.sh`, and the errors I saw yesterday in the log are now indeed gone.

I double-checked `CFW2<0>` using `FirmwareUpdater` manually again, and all four boards are displayed as is the case in the screenshot.

Then I increased the verbosity-level of `FirmewareUpdater.verify.head.sh` as suggested by you and it looks like it could open / talk to at least two devices. I didn't spot any more error messages, but I'm not sure if it is only two devices/boards it is ought to check.

I attach a copy of the command line log (the saved log-file in the log-directory is, for some reason, less verbose. 

Let me know how to proceed. 

Many thanks

[log.of.FirmwareUpdater.iCubHertfordshire01.verify.head_01.txt](https://github.com/robotology/icub-support/files/2812200/log.of.FirmwareUpdater.iCubHertfordshire01.verify.head_01.txt)

marcoaccame, 2019-01-30:
Hi @frank-foerster, it works as expected now. 

It can open the driver. I suspect that in your previous test you had the FirmwareUpdater GUI running which was holding the driver busy.

If you read the log you can see that the two boards (bll.head.CAN0:1 and mc4.head.CAN0:2) already have the correct firmware.

Now you must repeat the verify script for the other body parts.

I expect that you shall see everything OK apart that:
- the strain @ CAN2:1 (left foot) will be reported missing. It does not appear from your screenshot and there is no wrapper which uses it, but it is reported in the file `hardware/FT/right_foot_strain.xml`. If you don't find it then we can remove this xml file and edit the `network.iCubHertfordshire01.xml` to remove these boards.
- the strain @ CAN4:1 (rigth foot) ... the same as the one on left foot.
- some boards will need a firmware update (strain, mtb, mais).

And then ... run the `FirmwareUpdater.script.update.$BODYPART.sh` scripts you need.

frank-foerster, 2019-01-30:
Hi @marcoaccame, 
is it the line
   "SUCCESS!! the fw version on remote board matches the requirements"

that tells me that the installed version matches the latest version?  
So is it this line that I have to analyze in order to see whether I have to run an update? 

And what do mean by me not finding the strain@CAN2:1 board? How can I tell whether it's there or not?
With respect to the updating of the strain, mtb, and mais boards I assume that the *verify*.sh scripts will tell me with the above message, which ones need updating, right?

I'll run the verification scripts for the other body parts now and see what they report ... 

Thanks

marcoaccame, 2019-01-30:
yes, the line `"SUCCESS!! the fw version on remote board matches the requirements"` tells that the board does not need update.

about the missing strain boards, .... my mistake: they are `strain.left_leg.foot.CAN3:1` and `strain.right_leg.CAN4:14`.

about: `With respect to the updating of the strain, mtb, and mais boards I assume that the verify.sh scripts will tell me with the above message, which ones need updating, right?`: **yes**.

frank-foerster, 2019-01-30:
Hello again, 
I just ran updates of all body parts except for face - I assume that that is part of the head (?) - and upon manual inspection, via `FirmwareUpdater` it looks like all necessary updates have been performed. 

Hence I'll close this issue/request. I might send a pull-request to get the changed files committed soon.

Many thanks for the help!

closed 2019-01-30 by frank-foerster

frank-foerster, 2019-02-05:
Hi @marcoaccame , ok, thanks for this.