robotology/mesh-support #770: Replacing the wrist metal component with the newly designed component

Yeshasvitvs, 2019-01-31:
### Description of the modifications

In order to perform physical robot-robot interaction experiments we would like to replace the **old wrist component** (@julijenv I am not sure of the exact technical name of this part) shown below

[image]

with the **new component**

[image]

This will be changed on both the hands of `iCubGenova02` and `iCubGenova04`

I checked the availability of `iCubGenova02` with @arrenglover and it is free today to make the modification. @julijenv Please let me know how much time it would take to change it for each wrist.

Also `iCubGenova04` is available today to make the modifications.

Thank you

@ChiaraBartolozzi @arrenglover @S-Dafarra @GiulioRomualdi @DanielePucci

Yeshasvitvs, 2019-02-05:
Yesterday @julijenv tried to mount the new wrist part on the right hand of the green robot. We quickly discovered that the part is not accurately designed to fit correctly with the plastic part of the palm assemble. So, @julijenv filed one element to correctly fit the part. 

[image]

Then we double checked the design with michelle and the shortcomings in the part are due to a poor job from the machine shop. So, michelle requested to do the filing of the three other elements. This would take a few days and as soon as we have it we can proceed to mount it.

Yeshasvitvs, 2019-02-07:
We received the parts back from the machine shop but still, there are some minor changes to be done. So, we sent it back and hopefully, we will receive them tomorrow.

Yeshasvitvs, 2019-02-19:
Today we received two of the wrist components with the changes requested from the machine shop. Unfortunately, the last one was damaged as the drill bit is stuck inside it. So, the machine shop will be making a new component from scratch. This may take a week or more again.

julijenv, 2019-03-06:
Hi @Yeshasvitvs ,
I headr that the pieces could be arrived, tomorrow, I could confirmed and also try to put them on the purple robot.
lets cross fingers

Yeshasvitvs, 2019-03-07:
@julijenv thanks for the update!

Yeshasvitvs, 2019-03-16:
All the wrist components are correctly replaced! Thank you very much @julijenv and Michele! I will close the issue now!

closed 2019-03-16 by Yeshasvitvs