robotology/mesh-support #771: [Genova02] eye-lids drop down and cover eyes and are not controlled

arrenglover, 2019-02-01:
### Description of the failure

If the eye is controlled to tilt down and then back up, the eye-lids cover the eye and must be manually moved up again. 

### Detailed conditions and logs of the failure

There is currently no way to control the eye-lid motor in the yarpmotorgui, and I'm not sure if the motor control is exposed in some other place.

Currently the eye-lids have been removed as they cause problems doing the tracking demo.

julijenv, 2019-02-06:
Hi @arrenglover ,
I will need to add some files in order to operate correctly the eyelids, I dont know though if the peculiarity of the control of this motor when the eye tilt down allow the motor to compensatethe movement of the eye.

julijenv, 2019-03-18:
This shuld have been resolved by @marcoaccame .

closed 2019-03-18 by julijenv