robotology/mesh-support #773: [Genova02] face LEDs not installed

arrenglover, 2019-02-05:
### Description of the failure

The LEDs for the eyebrows and mouth have not been installed yet.

### Detailed conditions and logs of the failure

julijenv, 2019-02-06:
Hi @arrenglover,
I'll let you know, I check the availability of the pieces and I will mount them ASAP.

julijenv, 2019-06-27:
Hi @arrenglover ,
did we put the facial expressions on the robot, sorry i totally forget if we did.

arrenglover, 2019-06-27:
My understanding is that someone took the face just yesterday to install them. I don't know if it was completed or not.

julijenv, 2019-10-01:
@arrenglover , @ChiaraBartolozzi , @marco-monforte , @Iaxama ,
At the moment the LEDs adn wiring calbes are all set ready to be installed. we are just wiating to receive the support of the LEDs that get plugged into the face.
@Fabrizio69 , could you please give them an estimate of the date on which probably arrive those supports. thx in advance

Fabrizio69, 2019-10-02:
@julijenv the parts are available. You can request the material from Michelini.
I prepared the labels for the material withdrawal

julijenv, 2019-10-03:
they have ben mounted and now @davidetome is looking to make them working properly.

julijenv, 2019-10-07:
installed and verified. NOW WORKING.
thx @davidetome , @pattacini

closed 2019-10-07 by julijenv

pattacini, 2019-10-07:
There was a missing `--from` in the command line, thus the conf file wasn't found.