robotology/mesh-support #774: iSpeak/yarpdev sound playback corrupted

arrenglover, 2019-02-06:
### Description of the failure

When using the `iSpeak` and `yarpdev --device speech` modules to make the robot speak using the on-board speaker of the purple robot, the sound can be corrupted or cut-off before the sentence is finished.

### Detailed conditions and logs of the failure

iSpeak is correctly producing the sound file in the /tmp directory and if that file is played manually the sound is okay, i.e. using the `play` command from the command line.

when using the yarp device to play the sound automatically, it was originally played at the wrong sampling rate (from memory 32 kHz instead of 16 kHz) with a warning message that this was occuring. The device was changed from using the `aplay` bin to using the `play` bin by @vtikha which fixed the sound distortion. However the playback can sometimes be cut short, only playing the first 1-2 seconds of a sentence.

 - it seems like it is just a driver / software issue on the new boards in the icub-head.
 - it was playing more than just the first word (and quotation marks were used to make it play the full sentence)

arrenglover, 2019-10-09:
as far as I know this problem this exists, but no-one has needed to use the voice of the robot since the time of the issue. It will cause us a head-ache if the purple is needed for a demo on short notice.

traversaro, 2019-12-08:
cc @S-Dafarra if I remember correctly you started using the microphone for walking teleoperation demos, so you may be interested in this if you need to run the teleoperation on iCubGenova02 .

randaz81, 2019-12-08:
The description of the issue is unclear.

You say:

> if that file is played manually the sound is okay, i.e. using the play command from the command line.

and: 

>The device was changed from using the aplay bin to using the play bin by @vtikha which fixed the sound distortion. However the playback can sometimes be cut short, only playing the first 1-2 seconds of a sentence.

The two sentences are conflicting each other.
Additionally, in the second sentence, I cannot see any reference to yarpdev.
I think your problem is just a problem with the hardware or the playback Linux device on iCubGenva02.

Regarding the use frequency issue with aplay, you should indicate the complete command line you used.

- aplay belongs to the alsa package
- play belongs to the  sox package

these two executables accept additional parameters to set the properties of the audio stream, which are different between the two (see the manual).
If you are playing a _raw, headerless data,_  both of them cannot be used with just `aplay xxxx`, you need something more elaborate, such as `aplay -f S16_LE -c2 -r16000 xxxx`.  The fact that `play` is working and `aplay` is not working is probably just a coincidence due to different default settings.
Another possibility (more elegant!) is to save data not in the headerless format, but in one of commonly accepted standards, such as .wav or .mp3

Finally I recommend to rely on the latest yarp device drivers for playback and recording (i.e. audioPlayerWrapper/audioPlayerRecorder) which are not mentioned here (and thus not used, I think) especially for teleoperation purposes @traversaro @S-Dafarra. 
Please remember that a continuous, realtime audio streaming (like skype) is not possible using yarp. It implements only basic buffering and UDP communication. So received audio will be subject to a constant delay, determined by the size of the buffer.

triccyx, 2019-12-09:
This issue is quite old and at the moment @arrenglover doesn't remember exactly all the process.
I will try to reproduce the bug on a pc104 so we can understand if it is a SW/use/installation issue or just the HW on iCubGenva02 has problems.

pattacini, 2019-12-09:
@triccyx the problem showed up only on iCubGenova02, hence we'd need to carry out tests on that specific platform.

triccyx, 2019-12-09:
I have spoken with @S-Dafarra and he is quite sure that on their platform all is working fine.
https://github.com/dic-iit/component_wholebody-teleoperation/issues/197
So I will try on iCubGenova02 asap.

triccyx, 2019-12-10:
The following tests have been done :
1)On support PC connected to iCub purple:
` aplay xxx.wav` file for 1 minute 
The file has been correctly played --> No HW problem could be present on iCub purple
2)On iCub purple head:
portaudio plugin and dependence have been compiled and installed and the following command has been executed
`yarpdev --device portaudio --read --rate 48000 --allow-deprecated-devices`

The following error:
`ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
[INFO]Device number 0
[WARNING]Device "portaudio" is deprecated. Opening since the "allow-deprecated-devices" option was passed in the configuration.
[INFO]created device <portaudio>. See C++ class PortAudioDeviceDriver for documentation.
[WARNING]ServerFrameGrabber: error configuring interfaces for parsers
yarp: Port /grabber active at tcp://10.0.0.2:10236/
[ERROR]subdevice <portaudio> doesn't look like a framegrabber
yarpdev: ***ERROR*** driver <grabber> was found but could not open
An error occurred while closing the portaudio stream
Error number: -9988
Error message: Invalid stream pointer
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.`

As portaudio is deprecated, I suggest using (if necessary) the new plugin for this purpose instead of
trying to test again with deprecated portaudio @ra3vld @Nicogene

pattacini, 2019-12-10:
Cool 👍 thanks @triccyx 
I would proceed further with the final test using the latest plugin.

triccyx, 2019-12-11:
Thanks to @randaz81 the following tests have been executed on iCub purple:
1)Execute:
```
yarpdev --device speech --lingware-context speech --default-language en-GB --pitch 120 --speed 100
iSpeak --package speech-dev
```
connect ports
```
<from>/iSpeak/speech-dev/rpc</from>
<to>/icub/speech:rpc </to>
<protocol>tcp</protocol>
```
execute:

```
yarp write ... /iSpeach
"hello world hello world hello world hello world hello world hello world"
```
**the test has been successful**

2)Bypassing iSpeach, execute:
```
yarpdev --device speech --lingware-context speech --default-language en-GB --pitch 120 --speed 100
```
execute:

```
yarp write ... //icub/speech:rpc say "hello world hello world hello world hello world hello world hello world"
```
**the test has been successful**

In both cases the quality is bad

pattacini, 2019-12-11:
Nice @triccyx 👍 

With:

>In both cases the quality is bad

what do you mean exactly?

The original issue was opened because the sentence was cut out and didn't complete.
Is this still the case?

triccyx, 2019-12-11:
No, I mean that the voice quality is very poor. Perhaps this could be due to the microphone
low quality.  **The sentences have not been cut**. This issue can be closed.

pattacini, 2019-12-11:
Awesome 👍

closed 2019-12-11 by pattacini

linzdg, 2020-01-01:
> The description of the issue is unclear.
>
> You say:
>
>
> and:
>
>
> The two sentences are conflicting each other.
> Additionally, in the second sentence, I cannot see any reference to yarpdev.
> I think your [...]

>Hello, I found this problem I had met is similiar with your answer about realtime audio streaming. When I followed the tutorial with the test microphones,

>1.On the server type in a terminal
>icub@icubsrv:$ yarpdev --device portaudioPlayer
>2.On the head type in a terminal
>icub@icubsrv:$ yarpdev --device portaudioRecorder --start
>3.On the server type in a terminal
>icub@icubsrv:~$ yarp connect /audioRecorderWrapper/audio:o /audioPlayerWrapper/audio:i

>There was a delay of almost four seconds from the sound caught by the microphones on the head’s >ears to hearing on the server speakers. How to solve this problem? Is this normal?

pattacini, 2020-01-01:
@linzdg I've moved the issue you opened in YARP repository to this ticket system.
Here it is #904.