robotology/mesh-support #775: iCubErzelli03 - Controlling the facial expressions

ddetommaso, 2019-02-07:
### Description of the failure
The facial expressions work normally except for some conditions.

- The eye-lids system doesn't work. Sending commands to close the eyes (set eli sad) nothing happens.

- The robot starts with the LEDs activated even if the module is not running. I'd like to know if it's possible to disable the LED changing some configuration files.

- How it is possible to change LED colors (iCubErzelli03 has rfe board)  

Thanks

julijenv, 2019-02-12:
Hi @ddetommaso ,
unfortunately or should /i say fortunately for you the face of iCubErzelli03 is way more davcne than the old face that you were moving before. Now it is a real motor that activate the movement and so you can control it. The control is done by motorcontrol and not anymore with the little board attache to the head.
if you look into the yarpmotorgui you should see that the eyelid can be moved with a specific motor interface.
hope it helps,
for the colour changes I noticed to @valegagge to give you more details about it or where to look for info about it.

ddetommaso, 2019-02-15:
Hi @julijenv,
that sounds good! It works perfectly with the motorcontrol.
Still having more details about the facial expressions would be appreciated. 

Thank you!

valegagge, 2019-02-18:
Hi @ddetommaso ,
as @julijenv said, your robot has the Robot Face Expression (rfe) board, the new board for managing the expressions. In absence of new requirements, I developed its firmware in order to obtain the same behavior of the old board. 

If you have new needs, I can add them.
Please tell me if there are other requirements, in addition of above mentioned,  and their priority so I can develop them in the near future.
Thanks!

maggia80, 2019-02-18:
@MrAndrea

ddetommaso, 2019-02-27:
Hi @valegagge,

at the moment we're controlling the face expressions using the basic functionalities already available for the older robot'versions, but of course it would be nice to exploit more the new RFE.

I would suggest some features to implement (not urgently):
1. Adding the possibility to switch off the LEDs
2. Adding the possibility to modify the color of the expressions
3. Adding the possibility to have transitions (positional and color'gradients) in the changes between one facial expression and the other.

Thanks!

julijenv, 2019-10-09:
same request from issue #883

pattacini, 2019-10-25:
Hello gens, @ddetommaso @amaroyo @reafrancesco 

We'll deal with this feature request internally and find out a suitable place for it within our agenda of planned activities. Gonna let you know when this can be developed. Stay tuned.

closed 2019-11-08 by pattacini

Nicogene, 2020-06-29:
The updated version of documentation on facial expression is online: https://robotology.github.io/robotology-documentation/doc/html/group__icub__faceExpressions.html