robotology/mesh-support #776: iCubErzelli03 - Laptop ethernet card issue

ddetommaso, 2019-02-11:
### Description of the failure
The laptop provided with the iCubErzelli03 (Dell Latitude 5590) shows an issue with the ethernet card. The bandwidth is around 30Mbits/s instead of 1Gbits/s. 

### Detailed conditions and logs of the failure
`sudo lshw`

        *-network
             description: Ethernet interface
             product: Ethernet Connection (4) I219-LM
             vendor: Intel Corporation
             physical id: 1f.6
             bus info: pci@0000:00:1f.6
             logical name: enp0s31f6
             version: 21
             serial: e4:b9:7a:5d:5e:b8
             size: 10Mbit/s
             capacity: 1Gbit/s
             width: 32 bits
             clock: 33MHz
             capabilities: pm msi bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
             configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=3.2.6-k duplex=full firmware=0.1-4 ip=10.0.0.1 latency=0 link=yes multicast=yes port=twisted pair speed=10Mbit/s
             resources: irq:127 memory:ef200000-ef21ffff

`iperf -c 10.0.0.1`

------------------------------------------------------------
Client connecting to 10.0.0.1, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.2 port 54588 connected with 10.0.0.1 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  36.9 MBytes  30.8 Mbits/sec

Running the same test on another Ethernet interface of the same pc (usb-c) the bandwidth is around 1Gbits/s.

julijenv, 2019-02-12:
Hi @ddetommaso , could you please download a ubuntu live image and put it on a usb key - boot on the key and just use it - DO NOT INSTALL IT- test again iperf3 with either new ethernet cable and another server.
and let us know
thx in advance

ddetommaso, 2019-02-15:
Hi @julijenv,
thank you for your response. After replacing the eth driver with the latest version, the issue is gone.

closed 2019-02-15 by ddetommaso