robotology/mesh-support #779: Problem with the connection of  batteries to the robot backpack

kouroshD, 2019-02-21:
Yesterday during the walking experiment we face the problem that battries does not connect well to the robot backpack, so as result it fell down at a certain point while walking because of the power outage.

CC @MrAndrea  @maggia80

maggia80, 2019-02-22:
@kouroshD could you please give more details about the failure. Did you see the button on the backpack becoming red or switched off?

kouroshD, 2019-02-22:
If I am not wrong, it became switched off.  @GiulioRomualdi  @DanielePucci, correct me if I am wrong!

DanielePucci, 2019-02-22:
The problem was similar to https://github.com/robotology/icub-support/issues/651, but with the real battery instead of the dummy. Although we opened the contacts of the battery (similarly to what we did [here](https://user-images.githubusercontent.com/7060968/39890070-f5fe694c-5499-11e8-9f53-94ad022d560a.jpg)) we connection between the battery and the electronics was not stable, and the robot shut down during walking

DanielePucci, 2019-02-22:
As @MrAndrea was mentioning here https://github.com/robotology/icub-support/issues/651#issuecomment-414595143 it may be the time to re-think the connection between the battery and the robot

maggia80, 2019-02-26:
yes, I agree. We need to do a redesign of the battery-pack (and of the dummy battery-pack). Candidates for replacing the springs are: http://www.coda-systems.co.uk/catalog/Swift_Dock.html

DanielePucci, 2019-03-02:
We shall discuss about this relatively soon. We have some important demos by the end of the month and this issue is a little scaring

maggia80, 2019-03-14:
I have checked the battery and there are some missing parts: 
   
[image]

We need to put them back.

maggia80, 2019-03-14:
I requested them to Fabrizio.

DanielePucci, 2019-05-06:
@maggia80 @MrAndrea 
Sorry for bothering, but I do not remember the status of this. We did a tapullo that allowed us to temporarily work, right? (demo corriere della sera in Genoa if I remember correctly)

MrAndrea, 2019-05-20:
Thanks to Marcello S. we have new plastic parts and the battery which was fallen, is good again.
[image]

Regarding the _tapullo_, we used 2 plastic screws:
[image]

kouroshD, 2019-07-01:
I check the icub-green workstation, I have seen only one battery and there is no plastic screws with the battery. I am not sure if that is the fixed battery! Where can I find the fixed battery?

@DanielePucci @MrAndrea

MrAndrea, 2019-07-01:
@kouroshD the battery is still on my desk, you can come here and take it!

kouroshD, 2019-07-02:
Thank you @MrAndrea .

closed 2019-07-02 by kouroshD

traversaro, 2019-10-03:
Hi @kouroshD , if the problem was fixed, can you comment also on https://github.com/robotology/icub-tech-support/issues/651 ? Thanks a lot!